autowarefoundation/autoware_universe #6380: Trajectory topic is timed out while parking

ahmeddesokyebrahim, 2024-02-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Fixes #5539 

Currently, we are working on a task for fixing and improving Autoware logging system as Autoware creates a noisy debugging environment, making it challenging to identify and address genuine issues.

For more details about the issue, plan how to tackle it, and linked PR, please have a look [here](https://github.com/autowarefoundation/autoware.universe/issues/5539).

For this bug, when using planning simulator for a parking scenario, Autoware throws a warning message from `topic_state_monitor` saying 

```
[system.topic_state_monitor_scenario_planning_trajectory]: /planning/scenario_planning/trajectory topic is timeout. (operator()() at ~/autoware/src/universe/autoware.universe/system/topic_state_monitor/src/topic_state_monitor_core.cpp:155)
```

This [video](https://youtu.be/UgCDS1_il0c) demonstrates the issue.

### Expected behavior

` /planning/scenario_planning/trajectory ` should not be timed out while parking

### Actual behavior

` /planning/scenario_planning/trajectory ` is timed out while parking

### Steps to reproduce

1. Open planning simulator
2. Plan vehicle start pose and goal pose achieving parking scenario like in this [video](https://youtu.be/UgCDS1_il0c)
3. Let Autoware drive autonomously and check for the warning message for `topic_state_monitor`

You may use the PRs mentioned [here](https://github.com/autowarefoundation/autoware.universe/issues/5539#issuecomment-1869696363) for having more comfortable logging messages, if you want.

ahmeddesokyebrahim, 2024-02-29:
@mitsudome-r : I would highly appreciate if you can have a look and support in this issue, or let us know who is the current owner for `topic_state_monitor` that can support here.

closed 2024-03-28 by ahmeddesokyebrahim

kirinain, 2024-09-26:
@ahmeddesokyebrahim Hey there, I am also facing the same issue. Right after reach one goal pose, terminal starts showing the issue timeout, and vehicle doesn't plan trajectory for any other goal poses.
"[topic_state_monitor_node-11] [WARN] [1727344050.319146527] [system.topic_state_monitor_scenario_planning_trajectory]: /planning/scenario_planning/trajectory topic is timeout. Set ERROR in diagnostics.
"
Can you help fix this, I running it directly on docker.

PR #6696 (2024-03-28): fix(freespace_planner): fix parking trajectory errors and warnings when parking completed
## Description
fixes #6380 
fixes #6378 
fixes #6381 

## Related links

This PR is one of group of PRs that aim to fix Autoware logging system to achieve the goal of reducing excessive error and warning logs on Autoware launch. 

Part of:
- https://github.com/autowarefoundation/autoware.universe/issues/5539

## Related links
More details in the [issue](https://github.com/autowarefoundation/autoware.universe/issues/5539).

## Tests performed
- Before this PR : https://youtu.be/UgCDS1_il0c
- After this PR  [... cut]