autowarefoundation/autoware_universe #6401: feat(pointcloud_preprocessor): crop box filter update

YoshiRi, 2024-02-14:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Add new crop box feature to 

- handle multiple cropboxes
- enable to  choose cropbox frame

### Purpose

Currently in some settings, we use crop box filter muiltiple times to filter vehicle body and mirror reflections. Fro example:
[image]

This will increase computation burden due to

- publishing unnecessary pointcloud
- multiple times inspection for each point

So I think its better to assign cropbox parameter at once and do not duplicate crop box filter subsequently.

The other suggestion is; 
Since current crop box filter is processed in input_frame and crop box is often given in base_link or map frame, we can not filter other frame pointclouds.

### Possible approaches

- add new node to handle multiple cropboxes
- add cropbox frame parameters to control in which coordinate the input pointclouds should be filtered.

### Definition of done

Merge each feature PR

VRichardJP, 2024-02-15:
How does this perform against cascaded crop box filters?

If I understand correctly, this feature would do the same than cascaded filters, but without the intra process communication (which is supposedly fast). Is intra process delay that big?

YoshiRi, 2024-02-19:
@VRichardJP 

> Is intra process delay that big?

Hmm, It seems not so much(~10ms). 
In some projects, we confront of CPU resources problems so that it cause unintended system behavior.

So I wanted to 
- reduce unnecessary communication
- reduce process delay by doing same filtering and coordinate transformation