autowarefoundation/autoware_universe #6443: Planned trajectory is not aligned with the vehicle's heading when the car is initialized at the intersection.

Kim-mins, 2024-02-18:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I found that, when the ego vehicle is initialized at the intersection, the generated trajectory is sometimes vertical to the ego vehicle.

Here's the [[rviz video](https://drive.google.com/file/d/1EcFTcsuj909TbDl5y5Tx3rnvkgm3kU27/view?usp=sharing)] and [[launch.log](https://drive.google.com/file/d/1wXLN8RjoIwsLoQXrsaMAgjJ8Q8NUXYgq/view?usp=sharing)] file.

[image]

When I checked it several times, this does not occur every time, but sometimes.

### Expected behavior

I want the generated trajectory to be aligned with the ego vehicle's heading.

### Actual behavior

But the generated trajectory is not aligned.

### Steps to reproduce

I ran the [[**planning simulation**](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/)] with the prebuilt docker, version [[20240201](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/173888915?tag=humble-20240201-prebuilt-cuda-amd64)]
Pointcloud map: [[link](https://drive.google.com/file/d/1VygpEdoqm22Zmx9CM9go4UgnEXBcU89J/view?usp=drive_link)]
Lanelet2 map: [[link](https://drive.google.com/file/d/1sJPtpktjJorXPam7yZgCP2sfiMxW4kKQ/view?usp=drive_link)]

And I set the initial pose at the intersection, and set the goal point somewhere.

### Versions

I ran the planning simulation with the prebuilt docker, version [[20240201](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/173888915?tag=humble-20240201-prebuilt-cuda-amd64)]

### Possible causes

It seems the heading is not considered well for some positions of the intersection.

### Additional context

I also searched for the similar issues, and I guess https://github.com/autowarefoundation/autoware.universe/issues/6271 could be the similar one, that issue is regarding the `backward trajectory` though.

VRichardJP, 2024-02-20:
I think you face the same problem I face in https://github.com/autowarefoundation/autoware.universe/issues/6271#issuecomment-1920884945:

> Beside the backward trajectory which is really weird, I think there is a problem with the way `RouteHandler::planPathLaneletsBetweenCheckpoints` is implemented:
> 
> 1. find all lanelets containing ego pose. If the vehicle is on no lanelet, it looks for the closest lanelet instead (which feels a bit dangerous).
> 2. find closest lanelet to goal pose.
> 3. for each starting lanelet candidate, check if ego orientation is acceptable and try to generate a route the goal lanelet. Finally keep the route with the shortest length

In your case, the initial pose is located on 2 lanelets, so the route handler has to decide which one in correct. 

For 3, the hard coded yaw threshold is [90 degrees](https://github.com/autowarefoundation/autoware.universe/blob/c67bbe75dc7e2bad43aa8a75fe43d94c8f1c6f78/planning/route_handler/src/route_handler.cpp#L2138) (!!), so even a perpendicular lanelet like in your case may be included as candidate. Then, since the goal you have set is located on the right part of the map, the path passing by the perpendicular lanelet is likely shorter than the one passing by the correct lanelet. So eventually the route handler thinks the perpendicular lanelets is the correct one.

Kim-mins, 2024-02-20:
Thank you for clarifying the issue @VRichardJP!

I read your detailed description, and I agree that this issue is same with yours, because of the point 3.
> 3. for each starting lanelet candidate, check if ego orientation is acceptable and try to generate a route the goal lanelet. Finally keep the route with the shortest length

And I think this issue can be resolved if the idea you suggested could be applied. So I'll close this issue soon.

Thank you!

VRichardJP, 2024-02-20:
I would rather leave this issue open. The problem is worth fixing and your issue highlights it better than https://github.com/autowarefoundation/autoware.universe/issues/6271 (which focuses on a different problem.)

As a temporary fix, you could reduce this value https://github.com/autowarefoundation/autoware.universe/blob/c67bbe75dc7e2bad43aa8a75fe43d94c8f1c6f78/planning/route_handler/src/route_handler.cpp#L2138.

Kim-mins, 2024-02-20:
Thank you for reopening, and your patch suggestion!

Your patch seems to reduces yaw threshold and I think it would work! I'll try it.

And I'll check your issue too in my spare time, so that I can sync this issue with yours.

Thanks!

danielsanchezaran, 2024-03-08:
@Kim-mins could you check if this PR solves the issue on your side? https://github.com/autowarefoundation/autoware.universe/pull/6550

Kim-mins, 2024-03-14:
Sorry for late reply @danielsanchezaran. 

I tried to test your PR, but I failed to run the latest version now.
I mainly run Autoware with docker, so I think I can try after docker image with Autoware version 20240315 being uploaded.
I'll test and tell you the result as soon as possible.

Thank you!

Kim-mins, 2024-05-07:
Sorry for late reply @danielsanchezaran.

I've expected the latest docker image to be uploaded [here](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe) periodically, but it seems there's no update anymore..
So I ran [the planning simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/) of autoware(commit: [a79ef50498a6a301446f66e55e834fbc58e78362](https://github.com/autowarefoundation/autoware/commit/a79ef50498a6a301446f66e55e834fbc58e78362)) with [Open AD Kit](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/), but it seems that the issue remains on my side.. Could you please check it for me?

https://github.com/autowarefoundation/autoware.universe/assets/50267797/cc19a3e9-c671-4486-8e52-589872c3b762

idorobotics, 2024-05-09:
cc @oguzkaganozt

danielsanchezaran, 2024-05-10:
> Sorry for late reply @danielsanchezaran.
> 
> I've expected the latest docker image to be uploaded [here](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe) periodically, but it seems there's no update anymore.. So I ran [the planning simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/) of autoware(commit: [a79ef50498a6a301446f66e55e834fbc58e78362](https://github.com/autowarefoundation/autoware/commit/a79ef50498a6a301446f66e55e834fbc58e78362)) with [Open AD Kit](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/), but it seems that the issue remains on my side.. Could you please check it for me?
> 
>  Kazam_screencast_00050.mp4

Can you give me the autoware.universe hash please?

Kim-mins, 2024-05-10:
Sorry for bothering @danielsanchezaran and @idorobotics.

I thought that Autoware repository determines the version but it wasn't, and I was running the version around 20240223. I should have checked it more.
I tested with the latest version, and it works as intended.

Thank you for spending time for me! I'll close this issue soon.

danielsanchezaran, 2024-05-13:
@Kim-mins no problem, thanks for reporting the issue, I am glad it was solved.

closed 2024-05-13 by Kim-mins