autowarefoundation/autoware_universe #6452: behavior_planning_container dies because of guard_condition

beyzanurkaya, 2024-02-19:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

[image]

### Expected behavior

no crash

### Actual behavior

behavior_planning_container  dies

### Steps to reproduce

Give target position repeatedly

beyzanurkaya, 2024-02-19:
@takayuki5168 https://github.com/autowarefoundation/autoware.universe/issues/2684 is very similar to that one. do you have any idea what caused this?

maxime-clem, 2024-02-20:
It could be related to https://github.com/autowarefoundation/autoware/issues/4138
The "steps to reproduce" is similar (set a goal position again).

takayuki5168, 2024-02-28:
@beyzanurkaya Sorry, I don't have any idea.

beyzanurkaya, 2024-04-01:
ROS Issue: https://github.com/ros2/rclcpp/issues/2163

xiaoc88, 2024-06-06:
@maxime-clem @beyzanurkaya @takayuki5168 Hi,I also encountered the same problem. Has this issue been resolved? I found that using the command to publish in a loop easily reproduces this issue.
[image]

Command: ros2 topic pub --rate 1 /planning/mission_planning/goal geometry_msgs/msg/PoseStamped "{
  header: {stamp: {sec: 0, nanosec: 0}, frame_id: 'map'},
  pose: {position: {x: -171.8, y: 78.1, z: -1.27}, orientation: {x: 0.0, y: 0.0, z: 0.62, w: 0.78}}
}"
commit: ab8bbf56ecdacbabae89f56f4bc8b6ea1d7bf0df
[image]

-DCMAKE_BUILD_TYPE=Release
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
Ros2: Humble
[image]

beyzanurkaya, 2024-06-06:
@xiaoc88 I solved this problem by installing ROS2 from the source. This PR that tier4 opened for this issue solved the problem.

PR: [Fix SharedFuture from async_send_request never becomes valid (#2044) … · tier4/rclcpp@0189b7c](https://github.com/tier4/rclcpp/commit/0189b7c2146e558218b8f0775d8e006cd062787d)

FYI: @mitsudome-r @xmfcx

idorobotics, 2024-06-11:
>@xiaoc88 I solved this problem by installing ROS2 from the source. This PR that tier4 opened for this issue solved the problem.

PR: [Fix SharedFuture from async_send_request never becomes valid (#2044) … · tier4/rclcpp@0189b7c](https://github.com/tier4/rclcpp/commit/0189b7c2146e558218b8f0775d8e006cd062787d)

FYI: @mitsudome-r @xmfcx 
@beyzanurkaya can this issue be closed?

idorobotics, 2024-06-11:
> @xiaoc88 I solved this problem by installing ROS2 from the source. This PR that tier4 opened for this issue solved the problem.

@beyzanurkaya can this issue be closed?

closed 2024-06-14 by beyzanurkaya

luojiaxiang11, 2024-06-19:
> @maxime-clem @beyzanurkaya @takayuki5168 Hi,I also encountered the same problem. Has this issue been resolved? I found that using the command to publish in a loop easily reproduces this issue. [...]
 
Hello，have you solved this problem? I have encountered this problem as well, but it only occurs occasionally without a specific trigger scenario. Installing ROS2 from the source does not solve my problem.

maxime-clem, 2024-06-19:
@luojiaxiang11 are you using the custom `rclcpp` with this commit https://github.com/tier4/rclcpp/commit/0189b7c2146e558218b8f0775d8e006cd062787d ?

luojiaxiang11, 2024-06-19:
> @luojiaxiang11 are you using the custom `rclcpp` with this commit [tier4/rclcpp@0189b7c](https://github.com/tier4/rclcpp/commit/0189b7c2146e558218b8f0775d8e006cd062787d) ?

Yes,i using it.

maxime-clem, 2024-06-19:
Please share the log file (found in `~/.ros/log/{DATE-TIME-...}/`) showing the logs when the error happened.
If the error is the same let's reopen the issue.

luojiaxiang11, 2024-06-19:
@maxime-clem 
Errors are not exactly the same, but behavior_planning_container also died.Error message as shown in the figure.
[image]
The full log information as
[launch.log](https://github.com/user-attachments/files/15896376/launch.log)

maxime-clem, 2024-06-19:
Thank you for sharing the log. Unfortunately there is no information that could help identify the issue.
Which Autoware branch are you using ? If possible please open a new issue where you share details about your setup, and if possible a bag recording to try to reproduce the issue.

luojiaxiang11, 2024-06-19:
> Thank you for sharing the log. Unfortunately there is no information that could help identify the issue. Which Autoware branch are you using ? If possible please open a new issue where you share details about your setup, and if possible a bag recording to try to reproduce the issue.

Thank you for your suggestion ,and my branch is release/v1.0.

DavideGiancola, 2024-06-28:
hi, I still have the same problem as described. I upgraded to the version in the main today 28/June. Checking the version of ROS installed automatically in the container (with option --devel) the proposed modification in commit [tier4/rclcpp@0189b7c](https://github.com/tier4/rclcpp/commit/0189b7c2146e558218b8f0775d8e006cd062787d) is already present the problem persists. I have additionally observed that in one particular condition it does not occur: if the first goal is a parking space, the second will be successful. On the next goal the problem occurs again.

takayuki5168, 2024-07-03:
@DavideGiancola cc @kosuke55 @soblin 
If you come across the issue again, please let us know how to reproduce the issue.