autowarefoundation/autoware_universe #6487: Autoware sometimes drives over the velocity limit.

Kim-mins, 2024-02-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm running Autoware with Carla, and I found that **Autoware sometimes drives over the velocity limit**.
Here are the video, `ros2bag` file and `launch.log` file for more details: [[rviz]](https://drive.google.com/file/d/1XlNFVRtcpxF_T2LXbnochsBYYpheeyDv/view?usp=sharing) [[ros2bag]](https://drive.google.com/drive/folders/1xCU2oRWmpiVPYUQlHgLQjT7z06gNpaVl?usp=sharing) [[launch.log]](https://drive.google.com/file/d/1uD7feWHnc7yPO-XroQrhtcHdmdgblUX_/view?usp=sharing)

In the video, the `max_velocity` is set to `30 km/h`, but after the cornering, Autoware accelerates ignoring the velocity limit more than 10km/h.

### Expected behavior

I want Autoware to drive under the velocity limit.

### Actual behavior

But Autoware sometimes drives over the velocity limit.

### Steps to reproduce

1. Pull prebuilt Autoware docker image: https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/166911012?tag=humble-20240115-prebuilt-cuda-amd64
2. Run ros2bag file I provided above.

### Versions

- Carla: 0.9.13
- Autoware: [20240115 (prebuilt docker)](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/166911012?tag=humble-20240115-prebuilt-cuda-amd64)

maxime-clem, 2024-02-28:
I will try to reproduce the issue when I get CARLA running. In the meantime I have the following questions:
- Did you change the velocity limit somewhere or are you using the default parameters ?
- What map are you using ?
- Are you encountering this issue on a downward slope by any chance ?

Kim-mins, 2024-02-28:
Thank you for the response @maxime-clem!!

I used the map `Town01` of Carla, and here's the maps for Autoware: [[lanelet2]](https://drive.google.com/file/d/1vm9SvalJe7Bc9sh_8jK-0cPulVAov5uD/view?usp=sharing), [[pointcloudmap]](https://drive.google.com/file/d/1MvJlfjkw3LWFUs5hdE_KQ2_CQTusU8UN/view?usp=sharing).

Actually, the map **does not contain any information for changing speed limit**, so I changed the velocity limit manually(if not changed, it is fixed to 72km/h), following [the information from Carla](https://carla.readthedocs.io/en/0.9.14/python_api/#getters_17), by publishing a message to `/planning/scenario_planning/max_velocity` (I'm considering the conversion between `km/h` and `m/s` too).

Also, `Town01` does not have any slope, and the roads are totally flat.

Thanks!

maxime-clem, 2024-03-26:
@mraditya01 was not able to reproduce the issue when using the new CARLA interface he is working on.

Kim-mins, 2024-03-26:
Maybe I should report it when I meet this issue later again.

By the way, does the bridge supports the latest version too?
And will the bridge be merged into the main branch soon? How's your plan?

Thanks!

maxime-clem, 2024-03-26:
Yes we are planning to merge it into universe as soon as possible, but we are having some issues with the latest Autoware version. Once we can support the latest Autoware we will open the PR for review (https://github.com/autowarefoundation/autoware.universe/pull/6621).

Kim-mins, 2024-03-26:
Thank you for the quick response @maxime-clem!

That's a really good news! I'm currently running a past version(20240115), so I always worried about bothering you since the found issues could be outdated.
I hope the PR to be merged as soon as possible. I really look forward to it.

Thank you for spending time to this issue @maxime-clem @mraditya01! I'll reopen this issue when I find the same bug.

Thanks!

closed 2024-03-26 by Kim-mins