autowarefoundation/autoware_universe #6531: segmentation pointcloud fusion document has unrendered image

sezan92, 2024-03-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

this document is supposed to describe segmentation point cloud fusion. it does not have the rendered image. https://autowarefoundation.github.io/autoware.universe/main/perception/image_projection_based_fusion/docs/segmentation-pointcloud-fusion/ it seems the `segmentation_pointcloud_fusion.png` image is missing.

### Expected behavior

rendered image

### Actual behavior

no image

### Steps to reproduce

1. go to link https://autowarefoundation.github.io/autoware.universe/main/perception/image_projection_based_fusion/docs/segmentation-pointcloud-fusion/

### Versions

Any web browser

### Possible causes

the image is missing from `images` directory

badai-nguyen, 2024-03-04:
@sezan92 thank you a lot for your report, I will fix this bug.

sezan92, 2024-03-05:
> @sezan92 thank you a lot for your report, I will fix this bug.

@badai-nguyen let me know if i need to review as it seems only adding the image path

badai-nguyen, 2024-03-06:
@sezan92 Thank you for willingness to support. I create the PR https://github.com/autowarefoundation/autoware.universe/pull/6548 but cannot find you in reviewers list. Anyway, please let me known if you have any comment.

badai-nguyen, 2024-03-07:
@sezan92 
I close this as the issue is fixed at https://github.com/autowarefoundation/autoware.universe/pull/6548

closed 2024-03-07 by badai-nguyen