autowarefoundation/autoware_universe #6534: Autoware Evaluator Bug Report: Inappropriate Ego Car Behavior in Lane Changing and Collision Scenarios

hcn1519, 2024-03-02:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

While looking at various scenarios in Autoware Evaluator, I came across a couple of scenarios that Evaluator did not detect as errors but where the driving style seemed inappropriate in real driving situations.

### 1. Unsafe Lane Changing

- [UC-A-04-00007_003_case01_cmn:1](https://evaluation.ci.tier4.jp/evaluation/reports/5d7d61b3-abf7-5970-b30d-35c606ddf1e7/tests/994883fb-ee00-5afe-ae34-f56f12a4d104/97ebaefb-10ed-516c-ac38-05d10edc3056/1e0dc139-f3cf-56c8-bc76-29af2009bd76?project_id=awf)

The original purpose of the scenario is to detect whether the ego car avoids obstacles well. And it achieves its goal. However, if you look closely at the ego car's behavior, you can see that it is driving along the lines of the lane too long. In this case, a more appropriate behavior for the ego car would be to completely change lanes when it detects an obstacle on the trajectory and then return to the original lane after moving away from the obstacle.

[image]

### 2. Collision

- [UC-A-02-00003_PSim_1_case8_cmn:1](https://evaluation.ci.tier4.jp/evaluation/reports/5d7d61b3-abf7-5970-b30d-35c606ddf1e7/tests/7db0f3b8-a1c3-59dc-a7d1-7bddd1ee596b/a11152cc-f437-5de0-929d-d575384249f1/1ba36ebf-3f27-5385-bc48-fbcfb83fff60?project_id=awf)

This scenario is for detecting a collision, and `CollisionCondition` does not think ego car collides with another obstacle. However, this scenario should be perceived as a collision because the distance between the ego car and the obstacle is too close. When another car overtakes in front of the ego car while driving, if the distance between the ego car and the overtaking car gets too close, the ego car must slow down faster.

[image]

### Expected behavior

1. Unsafe Lane Changing: Appropriate behavior for the ego car would be to completely change lanes when it detects an obstacle on the trajectory and then return to the original lane after moving away from the obstacle.
2. Collision: When another car overtakes in front of the ego car while driving, if the distance between the ego car and the overtaking car gets too close, the ego car must slow down faster.

### Actual behavior

1. Unsafe Lane Changing: ego car is driving along the lines of the lane too long.
2. Collision: Distance between the ego car and the obstacle is too close.

### Steps to reproduce

You can replay the scenario if you have permission to access of the Autoware Evaluator.

1. Unsafe Lane Changing: [UC-A-04-00007_003_case01_cmn:1](https://evaluation.ci.tier4.jp/evaluation/reports/5d7d61b3-abf7-5970-b30d-35c606ddf1e7/tests/994883fb-ee00-5afe-ae34-f56f12a4d104/97ebaefb-10ed-516c-ac38-05d10edc3056/1e0dc139-f3cf-56c8-bc76-29af2009bd76?project_id=awf)
2. Collision: [UC-A-02-00003_PSim_1_case8_cmn:1](https://evaluation.ci.tier4.jp/evaluation/reports/5d7d61b3-abf7-5970-b30d-35c606ddf1e7/tests/7db0f3b8-a1c3-59dc-a7d1-7bddd1ee596b/a11152cc-f437-5de0-929d-d575384249f1/1ba36ebf-3f27-5385-bc48-fbcfb83fff60?project_id=awf)

satoshi-ota, 2024-03-28:
Hi @hcn1519 Sorry for late response.

> The original purpose of the scenario is to detect whether the ego car avoids obstacles well. And it achieves its goal. However, if you look closely at the ego car's behavior, you can see that it is [...]

For now, the lateral margin for UNKNOWN objects is set to 0.5m in [this](https://github.com/autowarefoundation/autoware_launch/blob/954145a38aeba3b7076f9006eedd34b1c76951f3/autoware_launch/config/planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner/avoidance/avoidance.param.yaml#L78-L81) yaml. So, if you want to make the ego behavior safer, please increase soft margin or hard margin. The module creates avoidance path so that the ego keeps the lateral distance `soft_margin + hard_margin`.

```yaml
        unknown:
...
          lateral_margin:
            soft_margin: 0.7                           # [m]
            hard_margin: -0.2                          # [m]
            hard_margin_for_parked_vehicle: -0.2       # [m]
...
```

satoshi-ota, 2024-03-28:
> This scenario is for detecting a collision, and CollisionCondition does not think ego car collides with another obstacle. However, this scenario should be perceived as a collision because the [...]

May I ask which module did you use in motion, `obstacle_stop_planner` or `obstacle_cruise_planner`? The former one doesn't have cut-in detection feature, so I think it couldn't deal with such a scenario. On the other hand, the latter one can detect such kind of vehicle and decrease the speed faster (it works based on object recognition result). If possible, please change the module and see the behavior.

yuntianyi-chen, 2024-04-01:
> Hi @hcn1519 Sorry for late response.
>
>
> For now, the lateral margin for UNKNOWN objects is set to 0.5m in [...]

The current scenario and condition are against traffic rules and safety concerns. Is this a kind of vulnerability or bug that could be improved (by changing the configurations) to enhance the safety and traffic-rule abidance in the future?

yuntianyi-chen, 2024-04-03:
> May I ask which module did you use in motion, `obstacle_stop_planner` or `obstacle_cruise_planner`? The former one doesn't have cut-in detection feature, so I think it couldn't deal with such a [...]

`obstacle_stop_planner ` was used in this test. Since it doesn't have the cut-in detection feature and was used as the default `motion_stop_planner_type` in the Planning module, it would cause dangerous issues such as collision during an obstacle cut-in scenario. Could we regard this as a potential bug or weakness that could be improved? @satoshi-ota

satoshi-ota, 2024-04-09:
> Could we regard this as a potential bug or weakness that could be improved?

Yes. It is a weakness we should improve. And, actually we made it possible to detect cut-in and decelete properly based on perception result in `obstacle_cruise_planner`.

satoshi-ota, 2024-04-09:
> The current scenario and condition are against traffic rules and safety concerns. Is this a kind of vulnerability or bug that could be improved (by changing the configurations) to enhance the safety and traffic-rule abidance in the future?

Yes. You can improve vehicle behavior by changing params. Please try to increase lateral margin params.

yuntianyi-chen, 2024-04-10:
Thanks for your confirmation! @satoshi-ota

closed 2024-06-05 by maxime-clem