autowarefoundation/autoware_universe #6535: A weird drivable area is generated for the merging lane.

Kim-mins, 2024-03-04:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with prebuilt docker(version [20240115](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/167294357?tag=humble-20240115-prebuilt-cuda)), and I found that a weird drivable area is generated for the merging lane, and it makes the ego vehicle stop forever.

Here's the [[rviz video](https://drive.google.com/file/d/1sPq4rY-itSb77J6mfD1msRACbzO-vJlh/view?usp=sharing)] and [[launch.log](https://drive.google.com/file/d/1fyx6XWjaskdl-E4ZniSamTxdM9jLJww6/view?usp=sharing)] files.

From the start of the rviz video, we can check that the shape of a drivable area is weird, and it gets worse as the vehicle tries to merge.

### Expected behavior

I think Autoware should drive to the destination if the routing was successful.

### Actual behavior

But Autoware fails to reach to the destination, even though the routing was successful.

### Steps to reproduce

1. You can check the situation with a ros2bag file [here](https://drive.google.com/drive/folders/11EaNdEUYOQSysDFl1hQxfHQ1mZb7rSlt?usp=sharing)
2. Or running Autoware with the following map files: [[lanelet2 map]](https://drive.google.com/file/d/1bvL4rbZXuoYGuw4wz94muxPeTWZIypQx/view?usp=drive_link) [[pointcloud map]](https://drive.google.com/file/d/1Uk8Kw0WCF_QcjzOgZN_3cr13fj1regN7/view?usp=drive_link)

### Versions

- Autoware: [20240115, prebuilt docker](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/167294357?tag=humble-20240115-prebuilt-cuda)
- Carla: 0.9.13
- OS (host): Ubuntu 20.04
- OS (docker): Ubuntu 22.04

### Possible causes

Sorry, I have no idea on this, but I assure that this issue is a special case, since I've seen Autoware worked well on the other maps with merging lanes(e.g., Town04).

satoshi-ota, 2024-03-28:
Hi @Kim-mins Sorry for late reply.

I saw attached video and thought the lanelet structure was weird. I tried to open your map but I couldn't. If possible, could you attach your lanelet2_map.osm?

[image]

Kim-mins, 2024-03-28:
No problem! thank you for the response @satoshi-ota!

Here's the map I'm using now: [[lanelet2_map](https://drive.google.com/file/d/1GG2g81ygBo2ioeFiuFkIjRuvqKxzCBN2/view?usp=sharing)]
Actually, I think it is not different with the map above, since I downloaded it from there..

I guess you have a problem on loading lanelet2 map on [Vector Map Builder](https://tools.tier4.jp/vector_map_builder_ll2/) with the message below:
[image]

I'm also suffering the issue on loading the lanelet2 map, and as far as I know, [JOSM](https://josm.openstreetmap.de/) is the only way to open the map.

Thank you!

Kim-mins, 2024-04-12:
Thank you for the response and the investigation @satoshi-ota!

I should have noticed there have to be the right bound.. sorry for bothering you.

After reading your response, I got two questions regarding the issue.

**1. Question on Autoware**
At first I thought this was a bug because I thought that, if routing was successful, Autoware could reach to the destination successfully (if there's no effect from surroundings), but it wasn't. Could you please share your opinion on this?
I agree that using bad lanelet2 map is more problematic, but I wonder if Autoware could emit routing failure beforehand.

**2. Question on lanelet2 map generation**
Do you know any proper tool for generating lanelet2(.osm) from opendrive file(.xodr)?
I've heard about the tools [assuremappingtools](https://github.com/hatem-darweesh/assuremappingtools) and [commonroad-scenario-designer](https://github.com/CommonRoad/commonroad-scenario-designer/tree/master), but I failed to run the former, and the latter does not generate the complete map(the generated map looks nice, but some errors here. Autoware does not work well with the map.).
So, if you have any experience on generating lanelet2 file from opendrive file for Autoware, could you please share the tool for me? Or if you successfully converted the map by using the two tools above, I want to ask you if you can share your experience.

Thanks a lot!

satoshi-ota, 2024-05-22:
> At first I thought this was a bug because I thought that, if routing was successful, Autoware could reach to the destination successfully (if there's no effect from surroundings), but it wasn't. Could you please share your opinion on this?
I agree that using bad lanelet2 map is more problematic, but I wonder if Autoware could emit routing failure beforehand.

Thanks for your comment. I guess we can validate the map and prevent output such a wierd output. In my opinion, we should validate the map when we create it instead of adding some feature in autoware. What do you think?
Actually, we TIER IV develop several validator for VMB and I think you can propose these kind of feature for map team if you want.

satoshi-ota, 2024-05-22:
> Do you know any proper tool for generating lanelet2(.osm) from opendrive file(.xodr)?
I've heard about the tools [assuremappingtools](https://github.com/hatem-darweesh/assuremappingtools) and [commonroad-scenario-designer](https://github.com/CommonRoad/commonroad-scenario-designer/tree/master), but I failed to run the former, and the latter does not generate the complete map(the generated map looks nice, but some errors here. Autoware does not work well with the map.).
So, if you have any experience on generating lanelet2 file from opendrive file for Autoware, could you please share the tool for me? Or if you successfully converted the map by using the two tools above, I want to ask you if you can share your experience.

Sorry. I don't know...

Kim-mins, 2024-05-23:
Thank you for the response @satoshi-ota!

I agree to your point. Just validating the map in offline seems better since there would be no problem if the map is validated once.
So, as you mentioned, the validation during the running seems not a good idea.

I'll close this issue soon.

Thank you!

closed 2024-05-25 by Kim-mins