autowarefoundation/autoware_universe #6539: Consolidate MRM related nodes into a single node

veqcc, 2024-03-04:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently, Minimum Risk Maneuver (MRM) is implemented across multiple nodes like the following:
(Pullover is a new feature still in Pull Request: https://github.com/autowarefoundation/autoware.universe/pull/5634)
[image]

You can refer to https://github.com/autowarefoundation/autoware.universe/issues/1856 to see when these MRM functions are introduced in autoware.universe.

From a point of safety and performance, I suggest consolidating these five nodes into a single node.

### Purpose

By consolidating these nodes, we aim to achive:
- improving reliability: a low probability of MRM failures since redundant ROS 2 node/topic/service will be removed.
- improving latency: a high performance of MRM behaviors on some emergency since:
  - we do not have to wait the MRM operation response from each node, or wait a timeout of some behaviors
  - we do not have to wait the topic/service to be scheduled by the ROS 2 executor

### Possible approaches

Implement each operators (including pull_over_manager) as a class in MRM Handler nodes:
[image]

If you want to add a new operator to MRM, just implementing a new class in MRM Handler is fine.

### Definition of done

 - [ ] consolidate these nodes into a single node

meliketanrikulu, 2024-05-09:
Hello @veqcc . Thanks for your effort. Is there any update related this issue.

veqcc, 2024-05-10:
Hi, @meliketanrikulu, thank you for your reminder.

Actualy, there will be a interface change in `vehicle_cmd_gate` from @isamu-takagi in the near future.
I will make some actions after that

veqcc, 2024-05-28:
I will close it.

closed 2024-05-28 by veqcc