autowarefoundation/autoware_universe #6546: Invalid path is generated when the point behind the vehicle is set as a goal

Kim-mins, 2024-03-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware version 20240115 with prebuilt docker on Carla, and I found that the invalid path is generated when the point behind the vehicle is set as a goal.

Here's the [[image]](https://drive.google.com/file/d/1ED0h6Jm6GpjTJBOwpXUBN0PlYLzFGXTB/view?usp=sharing) for the situation.
I set the goal point as the point behind the vehicle after spawning the ego vehicle, but the vehicle cannot move even though the routing was successful(I could press the button "AUTO" in RViz)

I also checked the [[launch.log]](https://drive.google.com/file/d/1zQFXdQhkWdSrRVnavG-oFghdIQmjw97h/view?usp=sharing) file, and I could see the log below,
```
1709580193.0063493 [component_container-73] [motion_utils] calcLongitudinalOffsetToSegment: Longitudinal offset calculation is not supported for the same points. Return NaN since no_throw option is enabled. The maintainer must check the code.
1709580193.0078657 [component_container-73] [ERROR] [1709580193.007519306] [control.control_validator]: predicted_trajectory size is less than 2. Cannot validate.
```
which states there's an error on calculating a longitudinal offset.

### Expected behavior

I want the vehicle to drive forward successfully to the goal point.

### Actual behavior

But the vehicle does not move at all.

### Steps to reproduce

1. Run Autoware of version [20240115, prebuilt docker](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/167294357?tag=humble-20240115-prebuilt-cuda) with the maps here: [[lanelet2 map]](https://drive.google.com/file/d/1vm9SvalJe7Bc9sh_8jK-0cPulVAov5uD/view?usp=drive_link), [[pointcloud map]](https://drive.google.com/file/d/1MvJlfjkw3LWFUs5hdE_KQ2_CQTusU8UN/view?usp=drive_link)
2. Spawn ego vehicle, and set the goal point as a point behind the vehicle which is on the same lane.

Or here's a [ros2bag file](https://drive.google.com/drive/folders/1RtaPGsjES0PuiNdmg2Tu30rgSXYPEU-C?usp=sharing) I recorded.

### Versions

- Autoware: [20240115, prebuilt docker](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/167294357?tag=humble-20240115-prebuilt-cuda)
- Carla: 0.9.13
- OS (host): Ubuntu 20.04
- OS (docker): Ubuntu 22.04

### Possible causes

According to the error message, I think the function related to the issue is this:
https://github.com/autowarefoundation/autoware.universe/blob/7ef61a96008eafc466c315f01a889540335787f1/common/motion_utils/include/motion_utils/trajectory/trajectory.hpp#L401

### Additional context

I thought [this](https://github.com/autowarefoundation/autoware.universe/issues/5440) and [this](https://github.com/autowarefoundation/autoware.universe/issues/5976) could be the same issue, but current issue occurs after fixing the first one, and for the latter one, I think it is not related to the backward path.

kyoichi-sugahara, 2024-03-05:
@Kim-mins 
Hello,
Thank you so much for the report!
In the current `start_planner` module the reported situation is not supported and explained [here](https://github.com/autowarefoundation/autoware.universe/blob/ebb4172992ef0a3d989a57af227d2545d404b63c/planning/behavior_path_start_planner_module/README.md/#L90) and implemented [here](https://github.com/autowarefoundation/autoware.universe/blob/ebb4172992ef0a3d989a57af227d2545d404b63c/planning/behavior_path_start_planner_module/src/start_planner_module.cpp)
If you have solution for this problem we are happy to discuss here and if you make PR me or other member can review the PR!

By the way what's the motivaiton for setting goal?
If my understanding is correct, backward driving on the road is necessary :thinking:

Kim-mins, 2024-03-11:
Thank you for the response @kyoichi-sugahara!!

Maybe I should have investigated more information for this issue.. sorry for bothering. I didn't know this is a known issue.
I'll let you know if I have any idea for this!

For the motivation, I just wanted to check if the vehicle could drive 'forward' and come back to the starting point well.
In my opinion, driving forward is correct, even though there's no other vehicles on the road, since that is the right direction of the road.
But I also think it's necessary to go backward when the vehicle is at a parking lot or alley, since there's no right direction.

maxime-clem, 2024-03-12:
Parking area can be represented in the lanelet map, in which case the freespace planner should take over and allow backward driving.
For the alley case, I imagine a case where a one lane alley is blocked and the ego vehicle needs to reverse to take another route. This case is currently not handled by Autoware.

kyoichi-sugahara, 2024-03-12:
@Kim-mins 
You're welcome and we all appreciate the feedback!! Thanks.
> I didn't know this is a known issue.

In my opinion, this is not the issue and it's not supported.

> But I also think it's necessary to go backward when the vehicle is at a parking lot or alley,

Same with maxime san's comment, for parking lot, parking lot information should be added in the current autoware and backward driving can be achieved.
Backward driving in alley case is not supported currently :bow:  Basiccaly direction of travel must be linked to a lane information.

In conclusion, autoware currently does not support the functionality to move in the opposite direction of the lane direction. However, if you feel there is a need for this, it would be a good idea to create a new issue regarding this matter and close this issue.(or update this issue) How does that sound?

Kim-mins, 2024-03-12:
Thank you for the kind responses @maxime-clem and @kyoichi-sugahara!

What I wanted to know from this issue was "why the vehicle cannot drive forward when the goal is at behind". So, I think it is ok for the vehicle not to drive backward.
And, for backward driving, after reading comment from you two, I got to know it is not supported for now and it is tricky to handle.
So, as @kyoichi-sugahara summarized, I agree to closing this issue, and open another issue for the feature for backward driving, but I think it is not too late to handle backward driving when needed, since I think the situation in alley(@maxime-clem noted above) would be not that frequent.

So.. may I close this issue for now?

ahsan155, 2024-05-13:
I just installed autoware and I am getting this same error when vehicle reaches the destination, how can I fix it? Do I need to change any property related to 2D goal pose?

maxime-clem, 2024-05-13:
@ahsan155 There is no workaround for this issue yet apart from splitting the route (instead of doing A -> C, first do A ->B, then B -> C).

ahsan155, 2024-05-13:
> @ahsan155 There is no workaround for this issue yet apart from splitting the route (instead of doing A -> C, first do A ->B, then B -> C).

thanks for the quick response. I am not even setting point behind vehicle as destination. I am just trying to move the vehicle on a straight path. why does it still gives error

[image]

maxime-clem, 2024-05-13:
These messages are just warning messages. This PR will prevent them from displaying in the terminal: https://github.com/autowarefoundation/autoware.universe/pull/6955
Otherwise I don't think there is any issue in your case.

meliketanrikulu, 2024-05-16:
Hello @Kim-mins . I think you can open discussion in [feature request](https://github.com/orgs/autowarefoundation/discussions/categories/feature-requests) section for this request. If it is convenient for you, we can close this issue this way. @maxime-clem

Kim-mins, 2024-05-17:
Hello @meliketanrikulu!

I just opened the feature request [here](https://github.com/orgs/autowarefoundation/discussions/4731)!
Please feel free to close this issue or keep it open.

Thanks!

closed 2024-05-17 by maxime-clem