autowarefoundation/autoware_universe #6555: obstacle cruise planner is setting zero velocity points behind the suddenly spawned obstacle in the trajectory

brkay54, 2024-03-06:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

In the `obstacle_cruise_planner` module of Autoware, an issue was observed where, upon the sudden appearance of a predicted obstacle on the planned route, the planner incorrectly sets a zero velocity for the trajectory segment behind the obstacle. This behavior deviates from expected handling, where adjustments are made to the trajectory in front of the obstacle. 

https://github.com/autowarefoundation/autoware.universe/assets/45468306/99de723f-0d1c-40b9-bde8-30d1cbb90ed3

As you can see in the video, `obstacle_cruise_planner` sets zero velocity points far away from the ego in trajectory.

The scenario was tested as part of the planning_control pipeline in the `reaction_analyzer` (please [see](https://github.com/autowarefoundation/autoware.universe/pull/5954) for more information.), highlighting the discrepancy in reaction time and trajectory planning when compared to `obstacle_stop_planner`. You can see the test results which are comparing motion_stop_planner_types in **Additional context** below.

### Expected behavior

When a sudden obstacle is spawned on the route, the `obstacle_cruise_planner` should dynamically adjust the vehicle's trajectory by reducing speed or stopping in front of the obstacle, ensuring safety and adherence to the planned path without creating impractical or unsafe zero-velocity segments behind the obstacle.

### Actual behavior

Upon the introduction of a PredictedObject in front of the vehicle, the `obstacle_cruise_planner` incorrectly sets a trajectory with zero velocity behind the sudden obstacle. This results in an unrealistic response to obstacles and may compromise the effectiveness of the planning_control pipeline. The warning message indicates a failure to interpolate the obstacle pose, suggesting a problem in trajectory adjustment logic.

### Steps to reproduce

1. Set motion_stop_planner_type `obstacle_cruise_planner` in default_preset.yaml
2. Create a route for the ego vehicle by using `planning_simulator.launch.xml`
3. Simulate the ego vehicle driving at approximately 40 kph.
4. Spawn a PredictedObject 15 meters in front of the moving vehicle.

### Additional context

Tests were made in some conditions:
- **Pipeline:** ``` [obstacle_cruise_planner or obstacle_stop_planner] -> scenario_selector -> motion_velocity_smoother -> planning_validator -> trajectory_follower -> vehicle_cmd_gate```
- Spawn obstacle if the distance between obstacle and vehicle is below 15 meters.
- Wait until all the topics are reacted to the object.

Reaction times for a sudden object:

**obstacle_cruise_planner:**
Node | Test - 1 | Test - 2 | Test - 3 | Test - 4 | Test - 5 | Test - 6 | Test - 7 | Test - 8 | Test - 9 | Test - 10 | Test - 11 | Test - 12 | Test - 13 | Test - 14 | Test - 15 | Min | Max | Mean | Median | StdDev
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
trajectory_follower | 62.8955 | 54.1983 | 63.3394 | 63.4726 | 83.6868 | 64.003 | 72.9368 | 73.4738 | 72.9419 | 72.9625 | 94.9728 | 52.9139 | 53.4631 | 63.7523 | 73.9409 | 52.9139 | 94.9728 | 68.1969 | 64.003 | 11.0857
vehicle_cmd_gate | 63.027 | 54.5334 | 63.7026 | 63.6133 | 83.8777 | 64.2321 | 73.0643 | 73.6687 | 73.141 | 73.2319 | 95.4233 | 53.282 | 53.6138 | 63.8961 | 74.2234 | 53.282 | 95.4233 | 68.4354 | 64.2321 | 11.1056
motion_velocity_smoother | 2050.22 | 2046.02 | 1954.43 | 1951.95 | 1955.34 | 2042.6 | 1952.83 | 2055.1 | 1951.57 | 1950.15 | 1943.56 | 1953.21 | 1958.31 | 1952.74 | 1955.94 | 1943.56 | 2055.1 | 1978.26 | 1954.43 | 42.5263
planning_validator | 2050.43 | 2046.29 | 1954.75 | 1952.31 | 1955.67 | 2042.87 | 1953.19 | 2055.42 | 1951.84 | 1950.42 | 1943.86 | 1953.54 | 1958.63 | 1953 | 1956.21 | 1943.86 | 2055.42 | 1978.56 | 1954.75 | 42.5089
obstacle_cruise_planner | 2146.54 | 2139.49 | 2149.69 | 2244.99 | 2244.96 | 2135.31 | 2142.88 | 2148.86 | 2057.46 | 2150.18 | 2238.32 | 2044.22 | 2045.75 | 2049.22 | 2052.21 | 2044.22 | 2244.99 | 2132.67 | 2142.88 | 69.3146
scenario_selector | 2146.68 | 2139.61 | 2149.84 | 2245.16 | 2245.27 | 2135.39 | 2143.09 | 2148.96 | 2057.59 | 2150.32 | 2238.45 | 2044.36 | 2045.88 | 2049.32 | 2052.34 | 2044.36 | 2245.27 | 2132.82 | 2143.09 | 69.3403

**obstacle_stop_planner use_predicted_objects: true:**

Node | Test - 1 | Test - 2 | Test - 3 | Test - 4 | Test - 5 | Test - 6 | Test - 7 | Test - 8 | Test - 9 | Test - 10 | Test - 11 | Test - 12 | Test - 13 | Test - 14 | Test - 15 | Min | Max | Mean | Median | StdDev
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
obstacle_stop_planner_use_predicted_objects | 81.8382 | 81.9391 | 83.8632 | 86.3468 | 85.1878 | 83.3875 | 82.3428 | 82.2222 | 84.5927 | 87.677 | 84.7681 | 85.0682 | 82.829 | 84.8964 | 87.0188 | 81.8382 | 87.677 | 84.2652 | 84.5927 | 1.78841
scenario_selector | 81.9589 | 82.1994 | 84.1792 | 86.5465 | 85.3413 | 83.5361 | 82.4551 | 82.4014 | 84.7381 | 87.8981 | 85.0498 | 85.1391 | 82.9631 | 85.0295 | 87.2563 | 81.9589 | 87.8981 | 84.4461 | 84.7381 | 1.80323
motion_velocity_smoother | 83.9044 | 84.0764 | 86.4751 | 88.8741 | 89.1653 | 85.5327 | 84.8375 | 84.6506 | 87.9979 | 90.7888 | 87.0837 | 88.8145 | 86.0696 | 87.6334 | 89.5126 | 83.9044 | 90.7888 | 87.0278 | 87.0837 | 2.08446
planning_validator | 84.1665 | 84.4794 | 86.8335 | 89.2254 | 89.3364 | 85.8473 | 85.2001 | 84.8592 | 88.3685 | 91.1991 | 87.4539 | 89.1279 | 86.392 | 87.9206 | 89.7858 | 84.1665 | 91.1991 | 87.3464 | 87.4539 | 2.0879
trajectory_follower | 107.754 | 87.8264 | 98.0843 | 108.317 | 107.828 | 108.065 | 88.6004 | 98.5973 | 97.9737 | 108.163 | 117.404 | 98.1266 | 107.861 | 117.814 | 98.005 | 87.8264 | 117.814 | 103.361 | 107.754 | 8.68018
vehicle_cmd_gate | 107.949 | 87.9589 | 98.2927 | 108.564 | 108.191 | 108.238 | 88.8766 | 98.8386 | 98.1012 | 108.342 | 117.555 | 98.4378 | 108.044 | 118.137 | 98.2119 | 87.9589 | 118.137 | 103.582 | 107.949 | 8.68948

**obstacle_stop_planner use_predicted_objects: false:**

Node | Test - 1 | Test - 2 | Test - 3 | Test - 4 | Test - 5 | Test - 6 | Test - 7 | Test - 8 | Test - 9 | Test - 10 | Test - 11 | Test - 12 | Test - 13 | Test - 14 | Test - 15 | Min | Max | Mean | Median | StdDev
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
obstacle_stop_planner | 86.8533 | 83.6418 | 82.7774 | 86.5559 | 85.7587 | 86.4881 | 87.1604 | 86.8174 | 86.0537 | 83.0285 | 84.5305 | 91.5692 | 85.014 | 84.4648 | 88.1908 | 82.7774 | 91.5692 | 85.927 | 86.0537 | 2.15022
scenario_selector | 87.135 | 83.7662 | 82.9173 | 86.7101 | 85.9345 | 86.6086 | 87.2528 | 87.0172 | 86.2334 | 83.1774 | 84.7482 | 91.7605 | 85.2001 | 84.5718 | 88.4958 | 82.9173 | 91.7605 | 86.1019 | 86.2334 | 2.17239
motion_velocity_smoother | 90.533 | 87.0821 | 86.2301 | 89.8849 | 88.4932 | 89.6786 | 89.3122 | 89.2584 | 88.2903 | 85.4724 | 86.8861 | 94.0338 | 87.3018 | 86.7852 | 91.0562 | 85.4724 | 94.0338 | 88.6866 | 88.4932 | 2.13953
planning_validator | 90.8006 | 87.4937 | 86.4362 | 90.1673 | 88.7433 | 90.0151 | 89.5907 | 89.5644 | 88.5552 | 85.8178 | 87.1727 | 94.4318 | 87.7218 | 87.0332 | 91.4043 | 85.8178 | 94.4318 | 88.9965 | 88.7433 | 2.15502
trajectory_follower | 121.686 | 101.488 | 111.798 | 121.782 | 102.148 | 111.794 | 92.0036 | 102.322 | 111.622 | 91.7295 | 102.278 | 111.838 | 91.8911 | 91.7672 | 101.677 | 91.7295 | 121.782 | 104.522 | 102.278 | 9.9253
vehicle_cmd_gate | 121.83 | 101.717 | 111.982 | 121.954 | 102.304 | 112.017 | 92.053 | 102.434 | 111.869 | 91.9243 | 102.436 | 112.019 | 92.0356 | 91.9892 | 101.878 | 91.9243 | 121.954 | 104.696 | 102.434 | 9.93526

takayuki5168, 2024-03-06:
@brkay54 
The obstacle_cruise_planner has minimum jerk and acceleration constraints for the stop based on the idea that the stop for the larger jerk and acceleration should be done by AEB.
Anyway, I think it would be better to implement a flag to disable them.

brkay54, 2024-03-06:
Hi, @takayuki5168 thank you for the quick answer. Then, it is expected behavior. 

> Anyway, I think it would be better to implement a flag to disable them.

After your answer, I looked at the code to find where these constraints were set and I found an if condition here:

https://github.com/autowarefoundation/autoware.universe/blob/81d53740a7f4673bc4c775bb60618708a8d7a94d/planning/obstacle_cruise_planner/src/planner_interface.cpp#L311-L325

I think this is already implemented. I tried to disable this parameter:

https://github.com/autowarefoundation/autoware_launch/blob/c03bd4bdb70117efffc328e5fe6e57426f169b3b/autoware_launch/config/planning/scenario_planning/lane_driving/motion_planning/obstacle_cruise_planner/obstacle_cruise_planner.param.yaml#L27

After disabling this parameter, `obstacle_cruise_planner` works as we expected. I tested again to be sure about it, here results:

Node | Test - 1 | Test - 2 | Test - 3 | Test - 4 | Test - 5 | Test - 6 | Test - 7 | Test - 8 | Test - 9 | Test - 10 | Test - 11 | Test - 12 | Test - 13 | Test - 14 | Test - 15 | Min | Max | Mean | Median | StdDev
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
obstacle_cruise_planner | 61.0829 | 59.778 | 64.9402 | 60.9953 | 64.5416 | 58.5924 | 61.2187 | 65.4465 | 58.2531 | 58.6755 | 61.2513 | 62.0713 | 62.39 | 58.3768 | 58.1821 | 58.1821 | 65.4465 | 61.053 | 61.0829 | 2.39521
scenario_selector | 61.2408 | 59.8985 | 65.2115 | 61.0872 | 64.6762 | 58.691 | 61.3608 | 65.5775 | 58.3899 | 58.7695 | 61.3561 | 62.2505 | 62.5848 | 58.5338 | 58.3239 | 58.3239 | 65.5775 | 61.1968 | 61.2408 | 2.41658
motion_velocity_smoother | 63.0515 | 62.0507 | 67.3218 | 63.1834 | 67.7317 | 60.7671 | 64.8069 | 68.4263 | 60.6618 | 60.9651 | 63.2338 | 65.5009 | 64.7324 | 60.982 | 60.5544 | 60.5544 | 68.4263 | 63.598 | 63.1834 | 2.62434
planning_validator | 63.3622 | 62.304 | 67.6128 | 63.4663 | 68.0695 | 61.0831 | 65.0504 | 68.7531 | 60.9425 | 61.2335 | 63.495 | 65.9171 | 65.1513 | 61.2585 | 60.8284 | 60.8284 | 68.7531 | 63.9018 | 63.4663 | 2.64667
trajectory_follower | 87.6072 | 77.7533 | 97.9342 | 86.9794 | 77.6086 | 87.9109 | 68.0986 | 97.6374 | 87.3483 | 77.1777 | 87.7728 | 97.4444 | 77.7068 | 77.1163 | 87.0026 | 68.0986 | 97.9342 | 84.8732 | 87.0026 | 8.51413
vehicle_cmd_gate | 87.838 | 77.9183 | 98.1974 | 87.1577 | 77.9118 | 88.085 | 68.2768 | 97.7869 | 87.5273 | 77.3806 | 88.0014 | 97.6075 | 77.9374 | 77.3294 | 87.1956 | 68.2768 | 98.1974 | 85.0767 | 87.1956 | 8.50809

I guess, it is already implemented, so I am closing the issue. Thank you @takayuki5168 -san!

closed 2024-03-06 by brkay54

takayuki5168, 2024-03-06:
Ah, right! I forgot that I already implemented it.