autowarefoundation/autoware_universe #6590: Separate the param.yaml file for ML model parameters and ML model handle node parameters

badai-nguyen, 2024-03-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently, the parameters releated to ML model and its execution node, including centerpoint and pointpainting are intertwined.
- This is a challenge for users as it's difficult to discern which parameters are adjustable and which are related to ML model structure or defined during training.
- Centerpoint/pointaining weights files are located [outside of autoware](https://github.com/autowarefoundation/autoware_launch/blob/0ce46ea580d6e782136157512079da9b0b30a56c/autoware_launch/launch/logging_simulator.launch.xml#L8) so it's is quite hard for users when trying param tuning for example [centerpoint's score_threshold](https://github.com/autowarefoundation/autoware.universe/blob/8bdb5422c682c1a2e9fa211e3c06be3a5f73ca55/perception/lidar_centerpoint/config/centerpoint.param.yaml#L16C5-L16C26) or [pointpainting's score_threshold](https://github.com/autowarefoundation/autoware.universe/blob/8bdb5422c682c1a2e9fa211e3c06be3a5f73ca55/perception/image_projection_based_fusion/config/pointpainting.param.yaml#L29C7-L29C28)

### Purpose

To clarify the fixed parameters which related to ML structure or training process and adjustable parameters which can modify by users. 

### Possible approaches

1. Separate ML model structure releated paramters and ML model execution node's parameters to different files
2. Locate ML model structure related param.yaml together with ML weights onnx files
3. Locate execution's node param.yaml file inside autoware (autoware_launch)

### Definition of done

1. Update launch and paramers in [universe](https://github.com/autowarefoundation/autoware.universe)
2. Update ML models/parameters [download scripts](https://github.com/autowarefoundation/autoware/blob/main/ansible/roles/artifacts/tasks/main.yaml) in autoware
3. Update parameters and launch in [autoware_launch](https://github.com/autowarefoundation/autoware_launch)

meliketanrikulu, 2024-05-09:
Hello @badai-nguyen . Thanks for your effort. Your PR 's looks merged. Is there any other active PR related this issue or can we close this issue.

badai-nguyen, 2024-05-10:
I close this issue since all neccessary PRs were merged!

closed 2024-05-10 by badai-nguyen