autowarefoundation/autoware_universe #6617: multi_object_tracker dies when unknown object has only one point in footprint which is 0,0,0

mebasoglu, 2024-03-14:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Sometimes, the input DetectedObjects message has an unknown object that has only one point inside the footprint field which is 0,0,0 point. When this message arrives, the node dies. Here is the terminal output:

```
[multi_object_tracker-63] terminate called after throwing an instance of 'boost::numeric::positive_overflow'
[multi_object_tracker-63]   what():  bad numeric conversion: positive overflow
[multi_object_tracker-63] *** Aborted at 1710407978 (unix time) try "date -d @1710407978" if you are using GNU date ***
[multi_object_tracker-63] PC: @                0x0 (unknown)
[multi_object_tracker-63] *** SIGABRT (@0x3e80001ef73) received by PID 126835 (TID 0x786d4e635f40) from PID 126835; stack trace: ***
[multi_object_tracker-63]     @     0x786d4cd07046 (unknown)
[multi_object_tracker-63]     @     0x786d4e042520 (unknown)
[multi_object_tracker-63]     @     0x786d4e0969fc pthread_kill
[multi_object_tracker-63]     @     0x786d4e042476 raise
[multi_object_tracker-63]     @     0x786d4e0287f3 abort
[multi_object_tracker-63]     @     0x786d4e4a2b9e (unknown)
[multi_object_tracker-63]     @     0x786d4e4ae20c (unknown)
[multi_object_tracker-63]     @     0x786d4e4ae277 std::terminate()
[multi_object_tracker-63]     @     0x786d4e4ae4d8 __cxa_throw
[multi_object_tracker-63]     @     0x786d4cf2f5a5 object_recognition_utils::getIntersectionArea()
[multi_object_tracker-63]     @     0x786d4cf61e91 object_recognition_utils::get2dIoU<>()
[multi_object_tracker-63]     @     0x786d4cf615de DataAssociation::calcScoreMatrix()
[multi_object_tracker-63]     @     0x786d4cee0ba2 MultiObjectTracker::onMeasurement()
[multi_object_tracker-63]     @     0x786d4cef2187 std::_Function_handler<>::_M_invoke()
[multi_object_tracker-63]     @     0x786d4cef20d2 _ZNSt8__detail9__variant17__gen_vtable_implINS0_12_Multi_arrayIPFNS0_21__deduce_visit_resultIvEEOZN6rclcpp23AnySubscriptionCallbackIN29autoware_auto_perception_msgs3msg16DetectedObjects_ISaIvEEESA_E8dispatchESt10shared_ptrISB_ERKNS5_11MessageInfoEEUlOT_E_RSt7variantIJSt8functionIFvRKSB_EESN_IFvSP_SH_EESN_IFvRKNS5_17SerializedMessageEEESN_IFvSW_SH_EESN_IFvSt10unique_ptrISB_St14default_deleteISB_EEEESN_IFvS14_SH_EESN_IFvS11_ISU_S12_ISU_EEEESN_IFvS1A_SH_EESN_IFvSD_ISO_EEESN_IFvS1F_SH_EESN_IFvSD_ISV_EEESN_IFvS1K_SH_EESN_IFvRKS1F_EESN_IFvS1Q_SH_EESN_IFvRKS1K_EESN_IFvS1W_SH_EESN_IFvSE_EESN_IFvSE_SH_EESN_IFvSD_ISU_EEESN_IFvS25_SH_EEEEEJEEESt16integer_sequenceImJLm8EEEE14__visit_invokeESL_S2B_
[multi_object_tracker-63]     @     0x786d4cf38fd3 rclcpp::Subscription<>::handle_message()
[multi_object_tracker-63]     @     0x786d4e8b77bc rclcpp::Executor::execute_subscription()
[multi_object_tracker-63]     @     0x786d4e8b7fbf rclcpp::Executor::execute_any_executable()
[multi_object_tracker-63]     @     0x786d4e8bf8b0 rclcpp::executors::SingleThreadedExecutor::spin()
[multi_object_tracker-63]     @     0x62377ef82510 main
[multi_object_tracker-63]     @     0x786d4e029d90 (unknown)
[multi_object_tracker-63]     @     0x786d4e029e40 __libc_start_main
[multi_object_tracker-63]     @     0x62377ef82c95 _start
[multi_object_tracker-63]     @                0x0 (unknown)
```

This issue is related with https://github.com/autowarefoundation/autoware.universe/pull/6597. On https://github.com/autowarefoundation/autoware.universe/pull/6597, the issue was solved for various known objects. However it still occurs for unknown objects.

### Expected behavior

The `multi_object_tracker` node should continue work even there are 0,0,0 footprint points.

### Actual behavior

The node dies.

### Steps to reproduce

The problem occurs when there is only `x=0`, `y=0`, `z=0` point inside the footprint of the unknown object from `/perception/object_recognition/detection/objects` topic.

I have a bag file to launch logging simulator, I can provide it if requested.

### Versions

- OS: Ubuntu 22.04 LTS
- ROS 2 Humble
- Autoware.Universe main

### Possible causes

The error is caused on this line:
https://github.com/autowarefoundation/autoware.universe/blob/ca46a8f34126cc9d1119c9cd11c80cda7de05578/perception/multi_object_tracker/src/data_association/data_association.cpp#L210-L211

Here, `measurement_object.shape.footprint.points` has only one point which is 0,0,0.

On https://github.com/autowarefoundation/autoware.universe/pull/6597, the minimum possible dimension was set as `0.3`.
Making also the footprint point `0.3, 0.3, 0.3` fixes the problem.

I tried adding this before the provided line above:

```C++
if (
            measurement_object.shape.type == autoware_auto_perception_msgs::msg::Shape::POLYGON &&
            area == 0 && measurement_object.shape.footprint.points.size() == 1) 
{
            measurement_object.shape.footprint.points.front().x = 0.3;
            measurement_object.shape.footprint.points.front().y = 0.3;
            measurement_object.shape.footprint.points.front().z = 0.3;
}
```

I am not sure if this is the best approach to fix but I will create a pull request to get review.

YoshiRi, 2024-03-16:
@mebasoglu 
Thanks for your report.
I think the fundamental problem with this issue is in the IoU calculation part.

Since we can not calculate IoU with line or point shape, which means the footprint length is 1 or 2, we should add escape sequence in `object_recognition_utils::get2dIoU` function.

Could you change the function rather than fixing multi_object_tracker_node?
https://github.com/autowarefoundation/autoware.universe/blob/e1d19d6c9d0fa86f85c0f7a9c92a0df2ef638684/common/object_recognition_utils/include/object_recognition_utils/matching.hpp#L66

cc:@yukkysaito @armaganarsln

mebasoglu, 2024-04-03:
Fixed in https://github.com/autowarefoundation/autoware.universe/pull/6701 .

closed 2024-04-03 by mebasoglu