autowarefoundation/autoware_universe #6659: Generated trajectory is far from the vehicle.

Kim-mins, 2024-03-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla, and I found a situation that, sometimes a generated trajectory is far from the ego vehicle.
Here's the video on RViz: [[rviz](https://drive.google.com/file/d/1wFA58Efv2TvpL0WJTQVw2U0_PCVOCsGm/view?usp=sharing)] and launch.log: [[launch.log](https://drive.google.com/file/d/1IFzKsMAFS-TTi8M6QVgij8TS1zn_aFpS/view?usp=sharing)]

As shown at the video, the ego vehicle is initialized well, but the generated trajectory is far from the ego vehicle, so that the vehicle cannot move at all.
I checked the position and some points of the trajectory, and they are really far each other:
```
# Position of the ego vehicle (/sensing/gnss/pose)
header:
  stamp:
    sec: 81
    nanosec: 705659884
  frame_id: map
pose:
  position:
    x: 1.674923330952879
    y: -79.10219146683812
    z: 19.401352405548096
  orientation:
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0

# Trajectory (/planning/scenario_planning/trajectory)
header:
  stamp:
    sec: 116
    nanosec: 505660403
  frame_id: map
points:
- time_from_start:
    sec: 0
    nanosec: 0
  pose:
    position:
      x: 235.05669529565205
      y: -2.557231487526552
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: -0.715651338833009
      w: 0.6984577018163102
  longitudinal_velocity_mps: 0.0
  lateral_velocity_mps: 0.0
  acceleration_mps2: 0.0
  heading_rate_rps: 0.0
  front_wheel_angle_rad: 0.0
  rear_wheel_angle_rad: 0.0
- time_from_start:
    sec: 0
    nanosec: 0
  pose:
    position:
      x: 235.05426392789735
      y: -2.6572019254111656
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: -0.715651338833009
      w: 0.6984577018163102
  longitudinal_velocity_mps: 0.0
  lateral_velocity_mps: 0.0
  acceleration_mps2: 0.0
  heading_rate_rps: 0.0
  front_wheel_angle_rad: 7.930140918932516e-13
  rear_wheel_angle_rad: 0.0
  ...
```

### Expected behavior

I wanted the ego vehicle to drive to the destination as usual.

### Actual behavior

However, the ego vehicle does not move at all, since the generated trajectory is too far from him.

### Steps to reproduce

Here's the ros2bag file for the reproduction: [[ros2bag](https://drive.google.com/drive/folders/1FQ2-7CbmurEdDXcpAN4pDG8GuWwTo6Rn?usp=sharing)]

### Versions

- Autoware: [20240115, prebuilt docker](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/167294357?tag=humble-20240115-prebuilt-cuda)
- Carla: 0.9.13
- OS (host): Ubuntu 20.04
- OS (docker): Ubuntu 22.04

takayuki5168, 2024-03-20:
@Kim-mins 
Please check if the map is correct or not. I don't know a good way though.
Also, please try various goal positions and see their behavior. (e.g. just in front of the ego)

Kim-mins, 2024-03-21:
Thank you for the response @takayuki5168!

I'm currently using the map by Hatem [[lanelet2](https://drive.google.com/file/d/1QZlolH6Zf2WDT9s_69dSoKklD6LeKAlD/view?usp=drive_link)] [[pointcloud map](https://drive.google.com/file/d/1Auk_jjOK8kTe38rjgZEzjgsWeUdl3egY/view?usp=drive_link)], and I've experienced that the ego vehicle could drive well with the given map. So I thought the map is ok.
Or, could you please tell me how to check the correctness of the map? I think I can try, but I don't know how..

I also tried other goal positions near to the original goal, and I found that the issue occurs when I set the goal point to the specific lane.
Here are the videos: [[video 1](https://drive.google.com/file/d/1ATnXuCuVtMAjh6Mix4JcxXwHJI_fckXl/view?usp=sharing)] [[video 2](https://drive.google.com/file/d/1vLKhan8i-QTyE5W0M2cGqJDsfyFC9h39/view?usp=sharing)]

In the first video, you can check that the trajectory is not valid sometimes, and in the second video, I tried the boundaries of the lane as a goal point, and it seems that the issue occurs when I set the goal point to the lane between two intersections.

So, I think this issue occurs when I set the goal point on the lane below:
[image]

But I cannot point out the root cause.. do you have any idea on this?

Thank you!

maxime-clem, 2024-03-21:
There could be a bug with that specific lanelet.
You can open the map with the [Vector Map Builder](https://tools.tier4.jp/vector_map_builder_ll2/) and see if you can find the issue.

Kim-mins, 2024-03-25:
Thank you for the response @maxime-clem!

I tried to open the map I'm using with Vector Map Builder, but it failed with the alarm below:
[image]
Instead, I tried with [JOSM](https://josm.openstreetmap.de/) and I could successfully open the map, but when I checked the lanelet, it is well connected with the adjacent lanelets.

I agree that it is hard to think the reason that makes the trajectory generation fail, except the reason that the shape of the intersection is not as usual.
So this seems the problem from my lanelet2 map file. 

Sorry for bothering and thank you for the kind responses @takayuki5168 @maxime-clem. I'll close this issue soon.

closed 2024-03-25 by Kim-mins