autowarefoundation/autoware_universe #6677: The tf published by `gnss_poser` competes with the one from `ndt_scan_matcher` and it creates a wrong tf structure tree as base_link has two parents.

marioney, 2024-03-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The `gnss_poser` package broadcasts a tf between `map_frame` and `base_frame`  tf in https://github.com/autowarefoundation/autoware.universe/blob/b44fb2014d241e656a56c0cf8d848866b33bd8b3/sensing/gnss_poser/src/gnss_poser_core.cpp#L196 and https://github.com/autowarefoundation/autoware.universe/blob/b44fb2014d241e656a56c0cf8d848866b33bd8b3/sensing/gnss_poser/src/gnss_poser_core.cpp#L376

When I use this package in combination with `ndt_scan_matcher` to give an initial position or to use the `regularization` option, the result is a duplicate of the tf between `base_frame` and `map`, one published by `ndt_scan_matcher` and the other by `gnss_poser`

If I use a different value for the `map_frame` parameter of `gnss_poser` then `base_frame` will have two parents and yield an unvalid tf tree structure. 

### Expected behavior

`gnss_poser` package should only broadcast this tf if explicitly required by the user via a ros param.

### Actual behavior

As describe above, `gnss_poser` package broadcast this tf all the time.

### Steps to reproduce

1. Configure and run `gnss_poser`
2. Run autoware using `ndt` as option for pose localization (Default)

### Versions

- OS: Ubuntu 22.04
- ROS 2: Humble
- Autoware: 2024.01

### Additional context

This is a duplicate of #6676 which was  created  Originally created from a post in a discussion  https://github.com/orgs/autowarefoundation/discussions/4088#discussioncomment-8875963

YamatoAndo, 2024-03-25:
`base_link` frame and `map` frame are directly connected.

[image]

marioney, 2024-03-25:
Thanks for the response.

Can you please share your `gnss_poser` configuration file?

I see a `gnss_base_link` frame I your tf tree, is it configured in there?

YamatoAndo, 2024-03-25:
Yes, I have configured `gnss_base_link`, just like in the latest Autoware.
https://github.com/autowarefoundation/autoware.universe/blob/3683131483dfdfc14f1490cb061c3589701a53ea/sensing/gnss_poser/config/gnss_poser.param.yaml

marioney, 2024-03-25:
But then I have two questions:

1. What's the relationship between `gnss_base_link` and `base_link`?
2. Why do you set the  `gnss_base_frame` to `gnss_base_link` and not `gnss_link` which is what's set on the [sensor_kit_description](https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/main/sample_sensor_kit_description/config/sensor_kit_calibration.yaml#L79C3-L79C12)

I think the tf between `map` and `gnss_base_link` is not needed, because `gnss_base_link` should be on-board the vehicle, and therefore have a static tf to `base_link`.

YamatoAndo, 2024-03-26:
> What's the relationship between gnss_base_link and base_link?

It's the estimated position of the base_link based on GNSS localization.

> Why do you set the gnss_base_frame to gnss_base_link and not gnss_link which is what's set on the [sensor_kit_description](https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/main/sample_sensor_kit_description/config/sensor_kit_calibration.yaml#L79C3-L79C12)

"gnss_link" refers to the antenna position of the GNSS receiver.

YamatoAndo, 2024-04-02:
@marioney Has your question been resolved?

YamatoAndo, 2024-04-09:
@marioney There is no problem in this report. So I'll close it.
Feel free to report anything else if needed.

closed 2024-04-09 by YamatoAndo

Yaoye1120, 2024-06-28:
> `base_link` frame and `map` frame are directly connected.
> 
> ![Screenshot from 2024-03-25 15-31-47](https://private-user-images.githubusercontent.com/13819566/316407791-0242500f-a009-43d5-b74c-8b147361cb4e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk1Njc4NDYsIm5iZiI6MTcxOTU2NzU0NiwicGF0aCI6Ii8xMzgxOTU2Ni8zMTY0MDc3OTEtMDI0MjUwMGYtYTAwOS00M2Q1LWI3NGMtOGIxNDczNjFjYjRlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI4VDA5MzkwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNhNmFiZTczNWQ5MWQ0ZDM5OGY4NzQ5YjUzOWY3ZTlmNzAyMjQ1NmY5ODg2M2NlODZiMGE4ZTMyODFlYTdkMjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Kz9SVtC-9QoJqqcT2euJQ4_IvLNLCV3dP8F0Ygziv2c)

hello, in this tf tree figure, which component published the tf from gnss_base_link to base_link ? 
I have launched gnss_poser.launch.xml, but terminator displays that "Please publish TF gnss_base_link to base_link" 
Sincerely !