autowarefoundation/autoware_universe #6708: build error in planning/behavior_velocity_crosswalk_module

mhmdrz-mirdamadi, 2024-03-28:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am using this project as a part of Autoware. When I try to build it, I face this error:
```
~/autoware/src/universe/autoware.universe/planning/behavior_velocity_crosswalk_module/src/scene_crosswalk.cpp: In member function ‘void behavior_velocity_planner::CrosswalkModule::applySafetySlowDownSpeed(autoware_auto_planning_msgs::msg::PathWithLaneId&, const std::vector<geometry_msgs::msg::Point_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Point_<std::allocator<void> > > >&, float)’:
~/autoware/src/universe/autoware.universe/planning/behavior_velocity_crosswalk_module/src/scene_crosswalk.cpp:882:23: error: ‘CROSSWALK’ is not a member of ‘autoware_adapi_v1_msgs::msg::VelocityFactor’ {aka ‘autoware_adapi_v1_msgs::msg::VelocityFactor_<std::allocator<void> >’}
  882 |       VelocityFactor::CROSSWALK);
      |                       ^~~~~~~~~
gmake[2]: *** [CMakeFiles/behavior_velocity_crosswalk_module.dir/build.make:118: CMakeFiles/behavior_velocity_crosswalk_module.dir/src/scene_crosswalk.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/behavior_velocity_crosswalk_module.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
```

### Expected behavior

To be built cleanly.

### Actual behavior

It fails to build.

### Steps to reproduce

`colcon build --packages-up-to behavior_velocity_crosswalk_module`

xuyuan-han, 2024-03-28:
I also met with this problem.

technolojin, 2024-03-29:
I think the reason is PR https://github.com/autowarefoundation/autoware.universe/pull/6670
@maxime-clem can you check if there is any missing update?

maxime-clem, 2024-03-29:
Thank you for reporting the issue. A fix PR (https://github.com/autowarefoundation/autoware.universe/pull/6711)  was merged so pulling the latest version should now allow you to build everything.
Please close the issue if the problem is fixed.

closed 2024-03-29 by mhmdrz-mirdamadi