autowarefoundation/autoware_universe #6709: Lazy object tracking makes Autoware collide with a pedestrian

Kim-mins, 2024-03-29:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla, and I found the situation that lazy object tracking makes Autoware collide with a pedestrian.
Here are the videos and the log of the situation: [[rviz](https://drive.google.com/file/d/1ARYERxpN-C15q0rdNUzIbv6SUr5ffa-a/view?usp=sharing)] [[frontview](https://drive.google.com/file/d/1wxtdmH1Ij9O6XRvyVELSMm8p6B5L-WcW/view?usp=sharing)] [[launch.log](https://drive.google.com/file/d/1YvYtQ5kpBdEmICjPCm2J9RHfC3MSbHwo/view?usp=sharing)]

In rviz video, it seems the pedestrian is sensed well by the pointcloud, but **he is not perceived as a pedestrian** until he invades the drivable area of Autoware.
The vehicle does not have enough stopping distance, and it eventually makes collision.

### Expected behavior

I expected the vehicle to detect the pedestrian well and no collision to happen.

### Actual behavior

The vehicle collides with a pedestrian.

### Steps to reproduce

Here's a ros2bag file for the reproduction: [[ros2bag](https://drive.google.com/drive/folders/1b7uSoY7J4mXwMAxjnhi2AWLKIkix5EnL?usp=sharing)]

### Versions

- Autoware: [20240115, prebuilt docker](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/167294357?tag=humble-20240115-prebuilt-cuda)
- Carla: 0.9.13
- OS (host): Ubuntu 20.04
- OS (docker): Ubuntu 22.04

technolojin, 2024-03-29:
Hi.

I saw your reporting video. It seems that the object is filtered out.

1. Lanenet filter
- Solution 1: expand your lanenet
- Solution 2: use `position_filter` instead. you may can select the launch option [DOCUMENTATION](https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/integrating-autoware/launch-autoware/perception/#tier4_perception_componentlaunchxml)

2. Initial object tracking performance
There was an update to adjust parameter for initial tracking. PR https://github.com/autowarefoundation/autoware.universe/pull/6343
Please try the latest version.

Kim-mins, 2024-04-01:
Thank you for the response @technolojin!!

I should have checked the PR too.. sorry for bothering you.
The video in the PR you provided, it seems the tracking works really well.
I'll try the latest version, and let you know if I face the same issue.

Thank you!

closed 2024-04-01 by Kim-mins