autowarefoundation/autoware_universe #6718: goal_planner (as well as start_planner) module of behavior_path_planner is not thread safe and we have a lot of room for refactoring to

soblin, 2024-03-31:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

# 5 non-thread-safe variables

## data race for planner_data_

[The PlannerManager runs each sub modules](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_planner/src/behavior_path_planner_node.cpp#L389) [inside the critical section of `BehaviorPathPlannerNode::planner_data_`](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_planner/src/behavior_path_planner_node.cpp#L353-L448), and the [`onTimer`](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_goal_planner_module/src/goal_planner_module.cpp#L104-L109) and [`onFreespaceParkingTimer`](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_goal_planner_module/src/goal_planner_module.cpp#L114-L120) belong to callback groups which are dependent on the callback group of [`bahavior_path_planner::run`](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_planner/src/behavior_path_planner_node.cpp#L167-L172), so the access to `GoalPlannerModule::planner_data_` from those timer callbacks are not thread-safe. What I found out is that when the onTimer tries to access `planner_data_->dynamic_objects`, it could be modified by the [msg callback `planner_data_->dynamic_objects = msg`](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_planner/src/behavior_path_planner_node.cpp#L774) at the same time because they are actually same shared_ptr object and GoalPlannerModule/behavior_path_planner does not share the mutex to it.

You may think "how could that be ? We are using shared_ptr<PlannerData> to share the resource across the threads, and if the consumer thread's access to shared_ptr<PlannerData>::dynamic_objects coincides with producer thread's update to shared_ptr<PlannerData>::dynamic_objects, then the latter one just repoints to new msg while keeping the former pointer, right ?"

One thing to note is that when behavior_path_planner calls [`PlannerManager::run()`](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_planner/src/planner_manager.cpp#L86) it passes its planner_data_ to each submodules via [`PlannerManager::setData()`](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_planner/src/planner_manager.cpp#L96-L97). As you can learn [from its implementation](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_planner_common/include/behavior_path_planner_common/interface/scene_module_interface.hpp#L251), this copy assignment increments the reference count of shared planner_data_ but [its internal shared_ptrs themselves](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_planner_common/include/behavior_path_planner_common/data_manager.hpp#L148-L163) **_are not copied_**. You can check that in the following snippet.

https://wandbox.org/permlink/Yxz8dNN3URSPxgbY

This means that although `GoalPlannerModule::planner_data_` and `behavior_path_planner::planner_data_` share the same resource, the PlannerData instance on the heap and its internal member shared_ptr variables are not shared and their reference count is still 1. So `GoalPlannerModule::planner_data_.dynamic_objects` and `behavior_path_planner::planner_data_.dynamic_objects` is **_identical object_**.

As the C++ standard says Read and Write access to different shared_ptr variables **are thread-safe even if they point to same resource**. In the following example `p1` and `p2` are different shared_ptr variables so we can freely manipulate them without mutex (We do not need to care about the use_count() because it is atomically added/subtracted).

```
shared_ptr<int> p1 = make_shared<int>();
auto p2 = p1;

// now p1.use_count() == p2.use_count() == 2

thread th1([&](){
shared_ptr<int> msg = make_shared<int>();
p1 = msg; // p1 is written
// now p1.use_count() 1
});

thread th2([&](){
cout << *p2 << endl; // p2 is read
// p2.use_count() == 1
});
```

This does not hold for our case because we are actually accessing to the same shared_ptr variable.

For the planner_data_ variable, the goal_planner and start_planner have to copy the PlannerData inside them to allow their custom timer thread access to it in thread-safe manner. Also we need to pay attention to the route_handler, because [it also holds shared_ptr variable in it](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/route_handler/include/route_handler/route_handler.hpp#L360-L370).
```
GoalPlannerModule::run() {
   {
       // copy latest planner_data_ to my_planner_data for the onTimer to use the latest information
       // also we copy the planner_data as value to increment the reference count as a new owner
       lock_guard<mutex> guard(goal_planner_mutex);
       my_planner_data = *planner_data_; // copy as value, not shared_ptr
       my_planner_data.route_handler = make_shared<RouteHandler>(*planner_data_->route_handler);
   }
  // then we do GoalPlanner stuff
  // in the following we should use planner_data_
}

GoalPlannerManager::onTimer() {
   PlannerData local_planner_data;
   {
   // here we copy my_planner_data to an another local data to minimize critical section
   lock_guard<mutex> guard(goal_planner_mutex);
   local_planner_data = my_planner_data;
   local_planner_data.route_handler = make_shared<RouteHandler>(my_planner_data.route_handler);
   }
   // then do onTimer stuff using local_planner_data
}
```

[... 5 lines cut ...]

The value of getCurrentStatus(), getPreviousModuleOutput() is updated through [setPreviousModuleOutput](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_planner/include/behavior_path_planner/planner_manager.hpp#L278) and [updateCurrentStatus](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_planner/include/behavior_path_planner/planner_manager.hpp#L286) in the callback group of `bahavior_path_planner::run` so the access from onTimer causes data race. We should put these variables into the planner_data as well.

## data race for occupancy_grid_map_ and last_previous_module_output_

- occupancy_grid_map_ is updated in the callback group of `bahavior_path_planner::run` through [GoalPlannerManager::updateData](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_goal_planner_module/src/goal_planner_module.cpp#L350-L369) so the access to it from [GoalPlannerModule::onFreespaceParkingTimer](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_goal_planner_module/src/goal_planner_module.cpp#L295-L324) through [GoalPlannerModule::isStuck](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_goal_planner_module/src/goal_planner_module.cpp#L1489-L1527) through [GoalPlanneManagerModule::checkOccupancyGridCollision](https://github.com/autowarefoundation/autoware.universe/blob/b5c29fc36bde2fe4d4a482d05b5fd375cfd00b96/planning/behavior_path_goal_planner_module/src/goal_planner_module.cpp#L1673-L1680) is not thread safe
- last_previous_module_output_ is also read in the main thread but updated in the onTimer thread

# minimize critical sections

Currently GoalPlannerModule has too many access to ThreadSafeData -- about 30-40 critical sections at least and maybe more than 100 if we count the net number of calls per one cycle. To be honest this is not a good design. I will post another refactoring roadmap.

### Expected behavior

[... cut ...] ta race at runtime **_if it happened_**. I think it is almost technically impossible to statically detect data race because behavior_path_planner modules are divided into different shared libraries, possibly preventing static tools to trace data paths.

I'll check if this option does not tell false positive data races from external libraries including rclcpp.

### Actual behavior

Sometimes behavior_path_planner dies from goal_planner logic.

### Steps to reproduce

There is no way to reproduce.

soblin, 2024-04-01:
Here is the code to mock the data race before/after the fix

```
sudo apt-get install libtsan0-amd64-cross
g++ -fsanitize=thread tsan.cpp
```

```
#include <chrono>
#include <iostream>
#include <memory>
#include <mutex>
#include <optional>
#include <thread>

using namespace std;

class PlannerData
{
public:
  PlannerData(const shared_ptr<int> ptr) : odometry(ptr) {}
  shared_ptr<int> odometry{nullptr};
};

int current()
{
  shared_ptr<PlannerData> planner_data = make_shared<PlannerData>(make_shared<int>(0));
  mutex planner_data_mutex;

  shared_ptr<PlannerData> my_planner_data = make_shared<PlannerData>(make_shared<int>(0));
  mutex my_planner_data_mutex;

  const int sim_duration = 10000;  // 10000ms = 10s
  int i = 0;
  const int tick = 10;
  thread odometryCallback([&]() {
    if (i > sim_duration / tick) {
      return;
    }
    {
      lock_guard<mutex> guard(planner_data_mutex);
      planner_data->odometry = make_shared<int>(i);
      ++i;
    }
    // 0.01秒ごとにplanner_data->odometryをインクリメントする（自己位置の更新の感じ）
    this_thread::sleep_for(chrono::milliseconds(tick));
  });

  int j = 0;
  const int tock = 500;
  thread onTimer([&]() {
    if (j > sim_duration / tock) {
      return;
    }
    // mutexなしで直接アクセスする
    // 自己位置を使った計算の模擬
    cout << *(planner_data->odometry) << endl;
    ++j;
    // 0.5秒ごとに読み出し
    this_thread::sleep_for(chrono::milliseconds(tock));
  });
  odometryCallback.join();
  onTimer.join();

  return 0;
}

int fixed()
{
  // これはOKっぽい？
  shared_ptr<PlannerData> planner_data = make_shared<PlannerData>(make_shared<int>(0));
  mutex planner_data_mutex;

  PlannerData my_planner_data(make_shared<int>(0));
  mutex my_planner_data_mutex;

  const int sim_duration = 20;  // 10s

  int i1 = 0;
  const double tick = 0.01;
  thread odometryCallback([&]() {
    while (i1 < sim_duration / tick) {
      {
        const auto msg = make_shared<int>(i1);
        lock_guard<mutex> guard(planner_data_mutex);
        planner_data->odometry = msg;
      }
      // 0.01秒ごとにplanner_data->odometryをインクリメントする（自己位置の更新の感じ）
      i1 += 1;
      this_thread::sleep_for(chrono::milliseconds(static_cast<int>(1000 * tick)));
    }
  });

  int i2 = 0;
  const double tack = 0.1;
  thread onGoalPlannerMain([&]() {
    while (i2 < sim_duration / tack) {
      {
        // 10Hzで動作
        // ここはPlannerManagerのスコープということで
        lock_guard<mutex> guard(planner_data_mutex);
        {
          // GoalPlannerのスコープ
          {
            // my_planner_dataをコピー
            lock_guard<mutex> my_guard(my_planner_data_mutex);
            my_planner_data = *planner_data;
          }
          // GoalPlannerのメイン処理はplanner_dataを使えば良い
          // 1/5でサブ処理
          if (planner_data->odometry) {
            const int a = *(planner_data->odometry);
          }
          this_thread::sleep_for(chrono::milliseconds(static_cast<int>(1000 * tack / 5.0)));
        }
        // 4/5で残りの処理
        i2 += 1;
        this_thread::sleep_for(chrono::milliseconds(static_cast<int>(1000 * 4.0 * tack / 5.0)));
      }
    }
  });

  int i3 = 0;
  const double tock = 0.2;  // 0.5秒ごとに動作
  thread onTimer([&]() {
    while (i3 < sim_duration / tock) {
      optional<PlannerData> local_planner_data{nullopt};
      {
        lock_guard<mutex> guard(my_planner_data_mutex);
        local_planner_data = my_planner_data;
      }
      if (local_planner_data) {
        cout << *(local_planner_data.value().odometry) << endl;
      }
      i3 += 1;
      this_thread::sleep_for(chrono::milliseconds(static_cast<int>(1000 * tock)));
    }
  });

  odometryCallback.join();
  onGoalPlannerMain.join();
  onTimer.join();

  return 0;
}

int main()
{
  fixed();
  return 0;
}
```

soblin, 2024-04-02:
NOTE

- [last_previous_module_output_ is written in onTimer](https://github.com/autowarefoundation/autoware.universe/blob/8f67bc90a88e427de59715240ad933cd67fc2855/planning/behavior_path_goal_planner_module/src/goal_planner_module.cpp#L292) and read from the main thread of GoalPlannerModule, so I will move this to ThreadSafeData
- [occupancy_grid_map_ is written in GoalPlannerModule::updateData through updateOccupancyGridMap](https://github.com/autowarefoundation/autoware.universe/blob/8f67bc90a88e427de59715240ad933cd67fc2855/planning/behavior_path_goal_planner_module/src/goal_planner_module.cpp#L139) but not updated on onTimer, so I will move this to GoalPlannerData

soblin, 2024-04-03:
Actually occupancy_grid_map_ ptr is thread safe, but it is no longer used in PullOverPlannerBase and its derived classes, so we can removed those members.

For goal_searcher_, its only functionality is to generate sorted goals, but once they are obtained we can pass them via ThreadSafeData. So we can replace the access to goal_searcher in the thread by thread_safe_data_/

idorobotics, 2024-04-17:
@soblin can this issue be closed as completed?

meliketanrikulu, 2024-05-09:
Thanks for your contribution @soblin . Is there anything missing regarding this issue? Can we close this ?

soblin, 2024-05-10:
We can close this issue. I will work on refactoring in the future PRs

closed 2024-05-10 by soblin