autowarefoundation/autoware_universe #6728: control_container dies because of the predicted_path_checker

beyzanurkaya, 2024-04-02:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

[image]

### Expected behavior

If the trajectory is empty, before using this variable in a calculation, it should check whether it has a point.

### Actual behavior

control_container crushes

### Steps to reproduce

https://github.com/autowarefoundation/autoware.universe/assets/32412808/2ad1ebe3-0785-4cfd-9697-e540eb6f0424

closed 2024-04-02 by brkay54

kyoichi-sugahara, 2024-04-03:
@beyzanurkaya 
Thank you so much for your quick report and addressing the issue. 

Just a minor comment: 
it is not preferable for the function that validates the output of control to cause node failures, leading to a complete shutdown of the container. 
As modified in the recent [PR](https://github.com/autowarefoundation/autoware.universe/pull/6435), it would be better to execute such validations outside the control_container in my opinion. 
Furthermore, Tier IV does not utilize the predicted_path_checker, it would be desirable to have a mechanism to enable/disable each function, similar to what is done in the planning module. 
The intention is not merely because Tier IV does not use it, but rather it would be preferable to have the flexibility to toggle features according to the varying needs of different users.

brkay54, 2024-04-03:
Hi @kyoichi-sugahara, thank you for the comment. The mission of the `predicted_path_checker` is sending emergency if there are any object in front of the vehicle. (Like `autonomous_emergency_braking` package.) 

> Furthermore, Tier IV does not utilize the predicted_path_checker, it would be desirable to have a mechanism to enable/disable each function, similar to what is done in the planning module.

Actually, user can enable or disable this module by using the launch file like AEB did (it is already disabled by default):
https://github.com/autowarefoundation/autoware_launch/blob/d475f0392bda0f1169cd1baabaf265698d736784/autoware_launch/launch/components/tier4_control_component.launch.xml#L8

kyoichi-sugahara, 2024-04-03:
@brkay54 
Thank you so much for the reply!

>The mission of the predicted_path_checker is sending emergency if there are any object in front of the vehicle. (Like autonomous_emergency_braking package.)

I see :thinking: Background of my proposal is to avoid the situation like control output is ok but container died because of later node's (e.g. control_validator) bug.  But if we can't accept such a situation, we can't use container...  
my idea toward this is not organized sorry. please ignore it for now.

> Actually, user can enable or disable this module by using the launch file like AEB did (it is already disabled by default):

Ops, I didn't notice this point sorry.
May be that's why I didn't notice this node exist. Thanks.