autowarefoundation/autoware_universe #6762: rviz2 Crashes with libGL Error on Ubuntu 22.04 Using CUDA 12.3 and Autoware Main Branch

jouta15123, 2024-04-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am encountering an issue where rviz2 crashes immediately upon launching with Autoware.universe running on the main branch. The specific error message is:
```
[rviz2-81] libGL error: failed to create drawable

``` 
This issue causes rviz to crash, preventing any further operation.

### Expected behavior

Expected Behavior
rviz2 launches successfully without crashing, allowing for normal operation and visualization within Autoware.

### Actual behavior

rviz2 crashes immediately upon launch with a libGL error: failed to create drawable error message.

### Steps to reproduce

1. Setup Autoware on the main branch with the specified environment.
2. Launch Autoware and initiate rviz2.
3. Observe the immediate crash of rviz2 with the libGL error: failed to create drawable error message.

### Versions

Autoware.universe: Main branch
Operating System: Ubuntu 22.04
CUDA version: 12.3
cuDNN version: 8.9.5.29-1+cuda12.2
TensorRT version: 8.6.1.6-1+cuda12.0

### Additional context

I previously had a working setup by downgrading CUDA to version 12.0. However, this workaround has suddenly stopped working, and I am now facing the same crashing issue with no change in the environment.

maxime-clem, 2024-04-08:
A few questions to help investigate the issue:
- Can you show the output of command `nvidia-smi` ?
- Does the issue happens when directly running the command `rviz2` ?
- Have you tried running inside docker ? Is the result the same ?

jouta15123, 2024-04-08:
Thank you for your reply.

- This is the output of nvidia-smi

```
nvidia-smi
Mon Apr  8 21:24:30 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.54.15              Driver Version: 550.54.15      CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3080 ...    Off |   00000000:01:00.0 Off |                  N/A |
| N/A   60C    P8             14W /   80W |    1016MiB /  16384MiB |     12%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      2093      G   /usr/lib/xorg/Xorg                            446MiB |
|    0   N/A  N/A      2447      G   /usr/bin/gnome-shell                           84MiB |
|    0   N/A  N/A      4608      G   ...seed-version=20240407-180234.263000        277MiB |
|    0   N/A  N/A    348374      G   ...yOnDemand --variations-seed-version        178MiB |
+-----------------------------------------------------------------------------------------+

``` 

- Regarding the execution of rviz2, it operates normally when I run the command directly in my environment.

- As for running inside Docker, autoware.universe works without any issues in a Docker environment.

maxime-clem, 2024-04-08:
> Regarding the execution of rviz2, it operates normally when I run the command directly in my environment.

The issue likely comes from the default Autoware rviz config. Maybe it contains some bad configuration, or maybe it loads some rviz panel that crashes on your machine.
If you are able to run `rviz2` directly, then you can try to manually configure it for your needs. This is a bit annoying, but it will at least allow you to use Autoware normally.
You can also try using different rviz configuration files to see if one works for you. Here are 2 config files you can try use with command `rviz2 -d PATH_TO_CONFIG_FILE`.
- https://github.com/autowarefoundation/autoware_launch/blob/main/autoware_launch/rviz/autoware.rviz
- https://github.com/tier4/scenario_simulator_v2/blob/master/openscenario/openscenario_interpreter/config/default.rviz

jouta15123, 2024-04-08:
Thank you for your response. You mentioned that modifying the autoware.rviz file might help in getting everything to work normally, but I'm not entirely clear on how to adjust the parameters effectively.

Could you please provide a more detailed explanation or some guidance on how to adjust the settings appropriately?

maxime-clem, 2024-04-09:
This page will give you an overview of how to configure Rviz yourself: https://docs.ros.org/en/humble/Tutorials/Intermediate/RViz/RViz-User-Guide/RViz-User-Guide.html#background
Otherwise please try the above 2 configuration files. If they work and do not cause any crash, you can use them and add any missing topics or panel yourself.
[image]
You can save/open config files in the `File` menu.
Also, please make sure you `source install/setup.sh` in the Autoware workspace before running `rviz2`, otherwise some topics and panels will be missing.

jouta15123, 2024-04-11:
Thank you very much for your detailed response.

I will attempt to implement the solution you've suggested. 
I deeply appreciate your assistance.

closed 2024-04-12 by jouta15123