autowarefoundation/autoware_universe #6787: The gpu_monitor publishes errors in diagnostics wtih "No message was set".

ito-san, 2024-04-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

On the latest version of Autoware, We are experiencing the following errors when running planning_simulator with system_monitor enabled.

- The gpu_monitor publishes errors in diagnostics wtih "No message was set".

### Expected behavior

The gpu_monitor publishes not errors in diagnostics.

### Actual behavior

The gpu_monitor publishes errors in diagnostics wtih "No message was set".

### Steps to reproduce

1. Build the latest version of Autoware.
    See [Autoware Documentation / Planning simulation / Installation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/scenario-simulation/planning-simulation/installation/).

2. Run planning_simulator with system_monitor enabled.
    ```console
    source ~/autoware/install/setup.bash
    ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/autoware_map/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit launch_system_monitor:=true
    ```

3. Run rqt_runtime_monitor.
    ```console
    ros2 run rqt_runtime_monitor rqt_runtime_monitor
    ```

ito-san, 2024-04-11:
`/usr/include/hwloc/nvml.h` is provided by Portable Hardware Locality project, not NVIDIA.
It seems that there is an issue with the setup.

ito-san, 2024-04-11:
Resolved the issue by installing the `libnvidia-ml-dev` package, which generated the `nvml.h` file under `/usr/include`.
```console
❯ find /usr -type f -name nvml.h
/usr/include/hwloc/nvml.h
/usr/include/nvml.h
```
[image]

ito-san, 2024-04-11:
```console
❯ dpkg -S /usr/local/cuda-12.3/targets/x86_64-linux/include/nvml.h
cuda-nvml-dev-12-3: /usr/local/cuda-12.3/targets/x86_64-linux/include/nvml.h
```
cuda-nvml-dev-{{ cuda__dash_case_cuda_version.stdout }} is better.

meliketanrikulu, 2024-05-09:
Hello @ito-san . Thanks for your contribution. Your PR looks merged. Can we close this issue?

ito-san, 2024-05-10:
Thank your for the follow up. I'm closing this issue.

closed 2024-05-10 by ito-san