autowarefoundation/autoware_universe #6815: Improve RViz UI - Adding 2D Minimap

KhalilSelyan, 2024-04-15:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

Leveraging popular and intuitive features from racing games, this task focuses on enhancing the rviz2 interface by adding a minimap and visual markers for checkpoints and goal points. These features will improve navigation and objective-tracking capabilities within simulations or real-world autonomous vehicle applications, making the interface more user-friendly and visually informative.

### Purpose

The primary goal of adding these features is to provide users with a more comprehensive and easy-to-navigate interface that integrates real-time spatial orientation and objective tracking, similar to a racing game.

### Possible approaches

1. Minimap Integration:
 - Implement a dynamic minimap in a corner of the rviz2 display that shows a top-down view of the surrounding area and the vehicle’s position within it.
 - The minimap should update in real-time and feature zoom in/out capabilities.

2. Checkpoint and Goal Visualization:
 - Develop visual markers for checkpoints and goals that are easily distinguishable from other markers, possibly using unique colors, shapes, or animations.
 - These markers should be visible on both the main display and the minimap, providing consistency across views.

### Definition of done

- [ ] A minimap has been successfully implemented and tested for real-time accuracy and responsiveness.
- [ ] Checkpoints and goals are clearly visualized on the minimap, with distinctive visual styles.

xmfcx, 2026-11-10:
- https://github.com/autowarefoundation/autoware_rviz_plugins/pull/49
- https://github.com/autowarefoundation/autoware_launch/pull/2017

https://github.com/user-attachments/assets/35a895c2-27fc-409a-863f-d832e974e3f7

It took 2.5 years to make it a part of autoware but it is done!

Thanks @KhalilSelyan !!!

closed 2026-11-10 by xmfcx

KhalilSelyan, 2026-11-10:
@xmfcx It looks great ! Thanks for the efforts on your side as well to port this to the latest version 🚀