autowarefoundation/autoware_universe #6816: centerpoint_tiny model causes crash

brkay54, 2024-04-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

While running e2e_simulator with AWSIM, I realized `centerpoint_tiny` model died. When I changed centerpoint model to `centerpoint` in `perception.launch.xml`, I couldn't see any problem.
https://github.com/autowarefoundation/autoware.universe/blob/9544e9c82c07a315daf75ec0d5e470fe15931fe0/launch/tier4_perception_launch/launch/perception.launch.xml#L39
Log:
```bash
[component_container_mt-1] [INFO 1713185581.007189900] [pointcloud_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<centerpoint::LidarCenterPointNode> (create_component_factory() at ./src/component_manager.cpp:133)
[component_container_mt-1] [I] [TRT] Loaded engine size: 0 MiB
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/default_ad_api/node/routing' in container '/default_ad_api/container'
[component_container_mt-1] [E] [TRT] 1: [runtime.cpp::parsePlan::314] Error Code 1: Serialization (Serialization assertion plan->header.magicTag == rt::kPLAN_MAGIC_TAG failed.)
[component_container_mt-1] [E] Failed to create context: Engine was not created
[component_container_mt-1] *** Aborted at 1713185582 (unix time) try "date -d @1713185582" if you are using GNU date ***
[component_container_mt-1] [I] Loaded engine from /home/berkay/autoware_data/lidar_centerpoint/pts_voxel_encoder_centerpoint_tiny.engine
[component_container_mt-1] PC: @                0x0 (unknown)
[component_container_mt-1] *** SIGSEGV (@0x8) received by PID 1678574 (TID 0x76bde7fff640) from PID 8; stack trace: ***
[component_container_mt-1]     @     0x76bcf87a7046 (unknown)
[component_container_mt-1]     @     0x76be09642520 (unknown)
[component_container_mt-1]     @     0x76bcf8587938 centerpoint::CenterPointTRT::CenterPointTRT()
[component_container_mt-1]     @     0x76bcf86f4b4a centerpoint::LidarCenterPointNode::LidarCenterPointNode()
[component_container_mt-1]     @     0x76bcf8724def rclcpp_components::NodeFactoryTemplate<>::create_node_instance()
[component_container_mt-1]     @     0x76be0a054961 rclcpp_components::ComponentManager::on_load_node()
[component_container_mt-1]     @     0x76be0a055925 std::_Function_handler<>::_M_invoke()
[component_container_mt-1]     @     0x76be0a05f2ab (unknown)
[component_container_mt-1]     @     0x76be09f2a2a6 (unknown)
[component_container_mt-1]     @     0x76be09f27c8a rclcpp::Executor::execute_service()
[component_container_mt-1]     @     0x76be09f27ff6 rclcpp::Executor::execute_any_executable()
[component_container_mt-1]     @     0x76be09f2f27a rclcpp::executors::MultiThreadedExecutor::run()
[component_container_mt-1]     @     0x76be09adc253 (unknown)
[component_container_mt-1]     @     0x76be09694ac3 (unknown)
[component_container_mt-1]     @     0x76be09726850 (unknown)
[component_container_mt-1]     @                0x0 (unknown)
[ERROR] [component_container_mt-1]: process has died [pid 1678574, exit code -11, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_mt --ros-args -r __node:=pointcloud_container -r __ns:=/ -p use_sim_time:=True -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7'].

### Expected behavior

Load CenterPoint node with `centerpoint_tiny` model successfully.

### Actual behavior

Node is dying.

### Steps to reproduce

Launch e2e_simulator with AWSIM.

xmfcx, 2024-04-15:
Could you share the results of the following commands:
```
nvidia-smi
screenfetch -n
```

brkay54, 2024-04-15:
```bash
berkay@victus:~$ nvidia-smi 
Mon Apr 15 16:33:07 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.54.15              Driver Version: 550.54.15      CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3060 ...    Off |   00000000:01:00.0 Off |                  N/A |
| N/A   61C    P5             30W /   30W |    4832MiB /   6144MiB |    100%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      1572      G   /usr/lib/xorg/Xorg                            382MiB |
|    0   N/A  N/A      2246      G   ...yOnDemand --variations-seed-version         53MiB |
|    0   N/A  N/A      3632      G   ...seed-version=20240414-180149.278000         32MiB |
|    0   N/A  N/A    218534      G   ...yOnDemand --variations-seed-version         23MiB |
|    0   N/A  N/A   1872199    C+G   ./AwsimLabs_V1.x86_64                        3405MiB |
|    0   N/A  N/A   1884169      C   ...p_components/component_container_mt        798MiB |
|    0   N/A  N/A   1885544      G   /opt/ros/humble/lib/rviz2/rviz2               107MiB |
+-----------------------------------------------------------------------------------------+

berkay@victus:~$ screenfetch -n
 berkay@victus
 OS: Ubuntu 22.04 jammy
 Kernel: x86_64 Linux 6.5.0-1019-oem
 Uptime: 6h 57m
 Packages: 3783
 Shell: bash 5.1.16
 Resolution: 1920x1080
 WM: i3
 GTK Theme: Adwaita-dark [GTK2/3]
 Icon Theme: Yaru-prussiangreen
 Font: Sans 10
 Disk: 441G / 733G (64%)
 CPU: 12th Gen Intel Core i7-12700H @ 20x 4.6GHz [81.0°C]
 GPU: NVIDIA GeForce RTX 3060 Laptop GPU
 RAM: 29433MiB / 63957MiB
```

xmfcx, 2024-04-15:
I'd guess that 6GB VRAM is not enough for both autoware's neural nets + AWSIM.

brkay54, 2024-04-16:
Sorry for the misunderstanding, it was my fault. After I removed the NNets files and re-installed, I couldn't observe this problem.

closed 2024-04-16 by brkay54