autowarefoundation/autoware_universe #6832: Concatenate node algorithm didn't work the same as it described

vividf, 2024-04-17:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

In the concatenate node description, it describes that if one sensor (pointcloud) fails, the node will still concatenate the remaining pointcloud. (see the last 5 pointcloud in the diagram shown below)
[image]

However, the algorithm implementation doesn't work as in the diagram. In the scenario when one of the `Pointcloud C `doesn't come, and the `Pointcloud D`, which already exists in the buffer, arrives at the node,  the timer will reset (start) with the period `timeout_sec`. In this case, the timeout won't happen in `t3` as the timer just starts again. Finally, Pointcloud E arrives and the current algorithm will concatenate `Pointcloud C, D, E`, and discard `Pointcloud A, B`. 

In some scenarios (image we have top, left, and right lidars) and we can control the publish time of each Lidar. We will set the most important pointcould (top) as the last-coming pointcloud. In this case, the current algorithm works fine because if the left and the right pointcloud `(Pointcloud A, B)` come and the top pointcloud (Pointcloud C) doesn't come. It is better to discard the left and the right pointcloud `(Pointcloud A, B)` as the concatenate pointcloud (without top pointcloud) might cause problems to the perception module. However, in this case, we probably need to add a parameter/algorithm that indicates which lidar is the main lidar, and when the main lidar doesn't arrive, we won't concatenate the pointcloud. 

reference: 
[https://github.com/autowarefoundation/autoware.universe/blob/main/sensing/pointcloud_preprocessor/docs/concatenate-data.md](url)

### Purpose

Provide a more robust algorithm for concatenating pointlcloud.  

### Possible approaches

This is the current algorithm of the concatenate node. 
[image]

Possible approach:
- add new parameters `main_pointcloud_topic`, which we don't concatenate the pointcloud if the pointcloud buffer lacks the `main_pointcloud`.
- Instead of resetting the timer to `timeout_sec` when receiving `Pointcloud D`, we do the concatenate if the main_pointcloud is in the buffer

Example
- If `Pointcloud C` is main_pointcloud, then do the same thing as the current algorithm does. 
- If `Pointcloud C` is not the main_pointcloud, we wait until the timeout then we publish. Or we receive `Pointcloud D` before the timeout, we also concatenate the pointcloud.

 

### Definition of done

The concatenate node can handle scenarios that are required.

vividf, 2024-05-09:
Possible solution diagram
[image]

palas21, 2024-06-25:
@vividf according to possible solutions without specifying the main_pointcloud, what happens if there are multiple pointcloud drops and when does the timer start, is it after getting the third pointcloud into the buffer?

vividf, 2024-06-25:
@palas21 
Thanks for your question.
The solution won't be the one described above. 
I will provide the decided solution later :)

vividf, 2024-07-12:
New design for concatenate node.

[image]

vividf, 2024-11-07:
This PR https://github.com/autowarefoundation/autoware.universe/pull/8300 solves the issue and will be merged soon.
Thus, I will close this issue.

closed 2024-11-07 by vividf