autowarefoundation/autoware_universe #6838: Implement feature extraction from projected point cloud and show

ataparlar, 2024-04-17:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

This task is the subtask of Point cloud feature extraction for LOAM based localization.
- https://github.com/autowarefoundation/autoware.universe/issues/6741
<hr>

For using them in the LOAM Based Localization, we need to export point cloud feature maps separately. Related issue:
- https://github.com/autowarefoundation/autoware/issues/4592

To implement feature extraction from a projected point cloud, the below task needs to be done:
- https://github.com/autowarefoundation/autoware.universe/issues/6837

### Purpose

Extracting the features (planar, edge...) from the generated point clouds.

### Possible approaches

- Use instantaneous point clouds for detecting the features.
- Use the full point cloud map for detecting the features.

### Definition of done

- [ ] Visualize the extracting process of the features in RViz as point clouds.
- [ ] Export the point cloud feature maps to `pcd` files.

ataparlar, 2024-04-26:
The tool can export corner feature point cloud map and surface feature point cloud map. Before exporting, it is visualizing both in the rviz. Right now, point cloud undistortion is not implemented. We wanted to see we can see the feature point clouds.

## RViz Visualization: 
### Corner Cloud
[image]

### Surface Cloud
[image]

## Exported .pcd:
### Corner Cloud
[image]

### Surface Cloud
[image]

meliketanrikulu, 2024-05-09:
Thanks for your effort @ataparlar . Could you share .pcd files too to examine

ataparlar, 2024-05-16:
I noticed that the projected image is not correct for some reason. The edge points are not extracted from the corners. They seems like just picking some random points from the point cloud. 

I debugged the projected point cloud's point angles with HSV color space for [loam_mapper](https://github.com/ataparlar/loam-mapper/tree/imu-deskew), original [LIO-SAM](https://github.com/TixiaoShan/LIO-SAM/tree/ros2) and [Velodyne VLP16 ROS2 driver](https://github.com/ros-drivers/velodyne/tree/humble-devel). The videos of them can found below:

Velodyne VLP16 ROS2 driver video:
https://youtu.be/NdwLPB2fWHc

Original LIO-SAM video:
https://youtu.be/sQA-Ilz-8BY

loam_mapper video:
https://youtu.be/tULkP0HXCQs

According to the comparison between those videos, in loam_mapper, the pixels that must be empty are very noisy. I am trying to solve that right now.

ataparlar, 2024-05-20:
The image_projection seems good right now. But the same error still exist. I am observing the `cloudInfo` message which is crucial for feature extraction. It keeps the point cloumn info, point range info, point start ring index and point end ring index for each point for each cloud. This thing is most probably filled wrong in the program. I am working on it. Here is the video of the last situation:
- https://youtu.be/9cJQO9Oyed4
The turquois points are the edge feature points extracted. They are so wrong obviously. I will keep here posted.

ataparlar, 2024-05-24:
Hey everyone. I solved the problem and I added point deskewing for feature extraction. Here is the demonstration video and an image of it:
- https://youtu.be/MbFCKkah0y4

[image]

I made the changes in `imu_deskew` branch and merged this branch into the master after it is done.

This subtask is done with the commit: 
- https://github.com/ataparlar/loam-mapper/tree/ec3697dcf67b0d4db0c21e648646685cf88df0f9

closed 2024-05-24 by ataparlar