autowarefoundation/autoware_universe #6844: Test Case Road Test Failed

cyn-liu, 2024-04-18:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Test Case Details:
[image]

### Expected behavior

The distance between the front of the ego and the set target after reaching the modified goal is less than D2[-2,2].

### Actual behavior

The distance between the front of the ego and the set target after reaching the  modified goal  is greater than D2[-2,2], which is approximately 4 meters.
test video:

https://github.com/autowarefoundation/autoware.universe/assets/104069308/fa7eca34-be3f-40be-81db-4c3ec6d99590

### Steps to reproduce

please refer to test case details.

### Versions

- OS: Ubuntu20.04 x86-amd64
- ROS2: Humble
- Autoware: release/2023.09

### Possible causes

1. Is the range setting of D2 parameter reasonable in this test case?
2. How to adjust Autoware parameters to make the modified goal meet the test requirements?

liuXinGangChina, 2024-04-18:
good afternoon，mist-san can you assign an engineer to take a look at this one @mitsudome-r

mehmetdogru, 2024-04-25:
@cyn-liu 

Could you please share the scenario and the map file?

cyn-liu, 2024-04-26:
@mehmetdogru 

this is our test scenario:
[image]

map files: [download link](https://drive.google.com/file/d/1NQ7BA9FuvmlOUmYXSZ1peMNOhDo8CpV9/view?usp=sharing)

mehmetdogru, 2024-04-26:
Thanks @cyn-liu 

Could you please share as well the scenario file created with [scenario editor](https://scenario.ci.tier4.jp/scenario_editor) if possible so that issue can be reproducible using [ssv2](https://github.com/tier4/scenario_simulator_v2) with your parameters specified in the scenario?

cyn-liu, 2024-04-28:
> Thanks @cyn-liu
> 
> Could you please share as well the scenario file created with [scenario editor](https://scenario.ci.tier4.jp/scenario_editor) if possible so that issue can be reproducible using [ssv2](https://github.com/tier4/scenario_simulator_v2) with your parameters specified in the scenario?

Thanks for your replay, but we don't have scenario file.
we use real vehicles for testing on the road.

mitsudome-r, 2024-05-07:
I have checked the original scenario definition.
I think the expectation is that you manually set the goal near the target vehicle within [-2 m , 2 m] in distance, which would be too close for Autoware vehicle to park without collision. Then, the Autoware should automatically modify the given goal by appropriately finding an open space in road shoulder and park to the modified goal.

cyn-liu, 2024-05-08:
> I have checked the original scenario definition. I think the expectation is that you manually set the goal near the target vehicle within [-2 m , 2 m] in distance, which would be too close for [...]

Thanks for your clarification, If the scenario definition is as you explained, then our road test  is successful.

meliketanrikulu, 2024-05-09:
> Thanks for your clarification, If the scenario definition is as you explained, then our road test is successful.

Hello @cyn-liu. In this case, can we close this task?

idorobotics, 2024-05-09:
> Thanks for your clarification, If the scenario definition is as you explained, then our road test is successful.

closed 2024-05-09 by idorobotics