autowarefoundation/autoware_universe #6851: Proposal for Integration and Contribution of Ouster Lidar and DURO GNSS to Autoware Repository

zeta-mobility, 2024-04-18:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

This issue concerns the integration and possible contribution of external sensors for Ouster Lidar and DURO GNSS, which are not currently available in the Autoware repository.

### Purpose

- **Ouster Lidar**: Documented as 'Autoware tested' on [Autoware's LIDAR documentation page](https://autowarefoundation.github.io/autoware-documentation/main/reference-hw/lidars/), yet no corresponding sensor integration code is available in the GitHub repository. We have developed our own integration code for the Ouster Lidar sensor and would like to discuss the possibility of contributing it to the repository.
- **DURO GNSS**: We have successfully integrated DURO GNSS with Autoware and are prepared to contribute our integration code to the software, despite it not being officially supported.

### Possible approaches

1. Review and discussion with maintainers about the interest and requirements for incorporating these ideas and code into the main Autoware repository.
2. If agreed, proceed with pull requests for each component.
3. Update documentation to reflect the addition of these new integrations.
4. Possible adjustments or extensions of CI to include these new components.

### Definition of done

- Clarification from maintainers regarding the contribution of Ouster Lidar and DURO GNSS integration.
- Acceptance of pull requests for both integration ideas, if approved.
- Documentation and CI updates to include these new integration approaches.

mitsudome-r, 2024-04-23:
> Ouster Lidar: Documented as 'Autoware tested' on [Autoware's LIDAR documentation page](https://autowarefoundation.github.io/autoware-documentation/main/reference-hw/lidars/), yet no corresponding [...]

Is you implementation anything different from https://autowarefoundation.github.io/autoware-documentation/main/reference-hw/lidars/?
Also, there is a work in Sensing and Perception Working Group to develop [unified lidar driver](https://github.com/tier4/nebula) for different vendors. Do you think your implementation fits to the design? (cc. @drwnz )

zeta-mobility, 2024-04-25:
Thank you for the response @mitsudome-r .

To clarify, we have not developed the driver itself; instead, we implement the code that converts the data received from the driver into the format required by Autoware's sensing pipeline. This transformation enables seamless integration of the data as an input to the sensing system. 

Thus, both the LiDAR(Ouster) and GNSS(Swift Duro) data were integrated not through the development of new drivers, but by modifying the data format to ensure compatibility with the existing system.

meliketanrikulu, 2024-05-09:
Hello @zeta-mobility . To utilize GNSS, you'll need to adjust your message type to [sensor_msgs/NavSatFix](https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/NavSatFix.html). I believe there's no need for additional additions as Autoware is presently compatible with this ROS default message type.

amadeuszsz, 2024-07-31:
Hi @zeta-mobility 
Are there any updates? 
> We have developed our own integration code for the Ouster Lidar sensor and would like to discuss the possibility of contributing it to the repository.

Recently Autoware updated the [point types](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-architecture/sensing/data-types/point-cloud/) and we have observed that some users are struggling with vendor drivers. If you're still interested in contributing, we'd like to encourage you to integrate Ouster sensors into [Nebula driver](https://github.com/tier4/nebula/tree/develop) 🚀

amadeuszsz, 2024-10-02:
Closing. Feel free to open this issue if you plan to contribute.

closed 2024-10-02 by amadeuszsz