autowarefoundation/autoware_universe #6863: [intersection] Incoming vehicle is ignored in some complex pattern

VRichardJP, 2024-04-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

We observed incoming vehicles are ignored by the intersection module in in some specific pattern. Here is a simple demonstration with the planning simulator:

https://github.com/autowarefoundation/autoware.universe/assets/18645627/b00d839b-05a4-4b9c-adc6-f1fdf9201582

As you can observe, an "intersection" wall appears when the dummy vehicle comes from below, but not when the vehicle comes from the left or the top. If I simplify the map by deleting all irrelevant lanelets, then incoming vehicles are always detected as expected:

https://github.com/autowarefoundation/autoware.universe/assets/18645627/0afa516f-b04f-493b-8b85-6d531846f023

I have tried to simplify the first lanelet map, but as soon as I remove 1 or 2 the problem disappear. I guess, the issue is related to the lanelet map being just a convoluted loop.

The 2 lanelet maps:

[lanelet2_map_intersection_fail.txt](https://github.com/autowarefoundation/autoware.universe/files/15061680/lanelet2_map_intersection_fail.txt)

[lanelet2_map_intersection_ok.txt](https://github.com/autowarefoundation/autoware.universe/files/15061682/lanelet2_map_intersection_ok.txt)

### Expected behavior

Intersection module detects all incoming vehicles

### Actual behavior

In the first case, vehicle coming from the top or left are systematically ignored

### Steps to reproduce

plannin simulator

### Versions

Ubuntu 22.04
main

xmfcx, 2024-05-15:
@soblin Sobue-san, are you actively working on this task? If not we can assign it to an Autoware Labs member since it may be failing some scenarios.

soblin, 2024-07-03:
@VRichardJP can you visualize the intersection module debug marker ? Also is it possible to upload the lanelet map file ?

VRichardJP, 2024-07-03:
@soblin the lanelet file is in the main report description.

soblin, 2024-08-15:
I've found out this issue is caused by the loop in the lanelet. I'm fixing the problem.

idorobotics, 2024-10-14:
> I've found out this issue is caused by the loop in the lanelet. I'm fixing the problem.

@soblin have you completed this task?