autowarefoundation/autoware_universe #6880: Traffic Light Indicator Weird Behavior

KhalilSelyan, 2024-04-24:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Traffic Light Indicator in the Rviz2 Overlay, flickers between indicators (green/red/yellow) because of sensor messages received ? planning seems to be doing the correct thing and following the indicator it should but not the indicator in the UI 

### Expected behavior

No flickering, just stable light indicator according to which one the planning is following

### Actual behavior

Signals received alternate between colors depending on which signal is first on the array of signals received in the message from the topic being subscribed to

### Steps to reproduce

1. Launch Autoware and AWSIM
2. Drive by a few traffic lights

### Versions

- OS : Ubuntu 22.04
- ROS: Humble

### Possible causes

The way the UI Indicator subscribes/displays the data is not the same as how the planning is being handled most likely

KhalilSelyan, 2024-04-24:
Ping: @xmfcx @miura @takayuki5168

maxime-clem, 2024-04-25:
I think we need to do the following:
- use the `traffic_signal_id` in the `TrafficSignal` messages to get the corresponding traffic lights in the map.
- calculate the traffic light longitudinal positions on the ego path.
- select the traffic light with the smallest longitudinal position (but ahead of the ego position).

Doing this requires adding many dependencies and topics to the rviz panel. If that's okay then I can implement it.

xmfcx, 2024-04-25:
@maxime-clem what about publishing this from where the planning code already does this as a debug message?

maxime-clem, 2024-04-25:
> @maxime-clem what about publishing this from where the planning code already does this as a debug message?

It may be better yes. I will try that first.

maxime-clem, 2024-04-26:
@KhalilSelyan I opened a PR with the change: https://github.com/autowarefoundation/autoware.universe/pull/6899
I cannot easily test it at the moment so please confirm if it fixes the chattering issue.

closed 2024-04-30 by xmfcx