autowarefoundation/autoware_universe #6915: `wheel_base` is not initialized for `AnalyticalJerkConstrainedSmoother` in `BehaviorVelocityPlannerNode`

LIIHWF, 2024-05-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

According to the code in `autoware.universe/planning/behavior_velocity_planner/node.cpp`, an `AnalyticalJerkConstrainedSmoother` is created by calling `onParam()` in the constructor of the node. 

However, nowhere is calling `setWheelBase`, leading the `base_param_.wheel_base` in the smoother to be uninitialized.

`base_param_.wheel_base` is accessed in `applySteeringRateLimit` method. The smoother in `BehaviorVelocityPlanner` can reach this method when executing `behavior_velocity_intersection_module` and calling `checkCollision` method in `scene_intersection.cpp`. The call chain is  `checkCollision -> util::calcIntersectionPassingTime -> smoothPath -> applySteeringRateLimit`.

This should be a BUG and can be corrected by initializing `wheel_base` somewhere.

### Expected behavior

 `base_param_.wheel_base` is initialized after `AnalyticalJerkConstrainedSmoother` being constructed in `autoware.universe/planning/behavior_velocity_planner/node.cpp (onParam)`

### Actual behavior

`AnalyticalJerkConstrainedSmoother` is constructed without any initialization in `autoware.universe/planning/behavior_velocity_planner/node.cpp (onParam)`

### Steps to reproduce

Check the call chain `checkCollision -> util::calcIntersectionPassingTime -> smoothPath -> applySteeringRateLimit` by seeing the sources:
- `autoware.universe/planning/behavior_velocity_planner/node.cpp` (`AnalyticalJerkConstrainedSmoother` is created in `onParam()`)
- `autoware.universe/planning/behavior_velocity_intersection_module/src/scene_intersection.cpp` (defines `checkCollision` called by `modifyPathVelocityDetail <- modifyPathVelocity`)
- `autoware.universe/planning/behavior_velocity_intersection_module/src/util.cpp` (defines `util::calcIntersectionPassingTime`)
- `autoware.universe/planning/motion_velocity_smoother/src/smoother/smoother_base.cpp` (defines `applySteeringRateLimit`)

soblin, 2024-05-08:
Thanks a lot, I will take a look at it

soblin, 2024-05-21:
https://github.com/autowarefoundation/autoware.universe/pull/7078

soblin, 2024-05-21:
The PR will be merged soon, thank you !

closed 2024-05-21 by soblin