autowarefoundation/autoware_universe #6920: What impact does autoware have on lidar point clouds in terms of bandwidth transmission?

The-wind-rises-2023, 2024-05-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When we tested on a dual ORin device, we found that when one ORin runs a semi-solid lidar driver, the point cloud frequency will decrease after the other ORIN starts autoware.

### Expected behavior

Normal point cloud frequency：10hz

### Actual behavior

Only 7hz

### Steps to reproduce

One of the dual orins runs the radar driver, and the other orin observes the point cloud frequency before and after starting autoware.

### Versions

Jetpack5.1.2 Docker autoware1.0 ros humble

knzo25, 2024-05-07:
@The-wind-rises-2023 
We experience that phenomena not only on Jetson devices but on other ECUs, desktops, etc.
Usually you need to tune your dds and network adapter settings.
This not really an autoware issue but general ROS/particular driver issue

idorobotics, 2024-05-07:
@knzo25 are you able to share your DDS parameter and/or network settings?

meliketanrikulu, 2024-05-09:
Hello @idorobotics . Did you tested with new DDS configs? Is [this](https://autowarefoundation.github.io/autoware-documentation/main/installation/additional-settings-for-developers/) documentation fixed your problems ?

idorobotics, 2024-05-09:
> Hello @idorobotics . Did you tested with new DDS configs? Is [this](https://autowarefoundation.github.io/autoware-documentation/main/installation/additional-settings-for-developers/) documentation fixed your problems ?

cc @knzo25

knzo25, 2024-05-10:
I think you meant to mention @The-wind-rises-2023

vividf, 2024-07-23:
@The-wind-rises-2023 
Kindly ping.
If the problem is solved, I would like to close this issue.

closed 2024-08-16 by vividf