autowarefoundation/autoware_universe #6937: Occasional Black Screen in RViz on Autoware Launch

sgNicola, 2024-05-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

An intermittent issue where RViz displays a black screen upon launching Autoware.Universe, despite Autoware state processes (e.g., initialization, waiting for route) appearing normal. This issue occurs even when GPU utilization remains typical, suggesting that the problem may not be directly related to hardware resource availability.
[image]
[image]
[image]

### Expected behavior

RViz should display the appropriate visualizations and interface elements consistently upon every launch of Autoware.Universe without fail.

### Actual behavior

Sometimes, when Autoware.Universe is launched, RViz fails to display its usual interface and visualizations, instead showing a black screen. This occurs while the Autoware state processes indicate normal operations, and there is no significant change in GPU utilization.The universe is launched, and RViz fails to display its usual interface and visualizations, instead showing a black screen.

### Steps to reproduce

1. Launch Autoware.Universe.
2. Observe the RViz window after launching.
3. Note any occurrences of the RViz window displaying a black screen instead of the expected content.

### Versions

-OS: Ubuntu 20.04
-ROS2: ROS2 galactic
-Autoware: Autoware.universe galactic

maxime-clem, 2024-05-08:
Just to confirm:
- are you running Autoware from source or from a Docker environment ?
- if you close rviz2 it should open again, does it fix the issue ?
  - Actually I am not sure if `<arg name="rviz_respawn" default="true"/>` exists in the galactic branch.
- is there any error message displayed by rviz ? If you are not sure, please share a log from when the issue occured (you can find the log in `$HOME/.ros/log/{DATE-TIME}/launch.log`.

sgNicola, 2024-05-08:
Thank you for your comment, 
1. I am running Autoware from source. 
2. It will fix the issue when I restart autoware. I will try to reopen rviz next time.
3. The log message not display any error message about rviz.
[launch.log](https://github.com/autowarefoundation/autoware.universe/files/15245133/launch.log)

sgNicola, 2024-05-15:
Hi, I encountered the issue again. I reopened rviz2, and it fixed the problem.

KhalilSelyan, 2024-05-27:
@maxime-clem this also happens on humble as well in main branch, with respawn just closing rviz client when it reopens it works fine but the black screen does occasionally show as well in main brain