autowarefoundation/autoware_universe #6938: obstacle pointcloud based validator node Incorrectly removes the small DetectedObjects (traffic cone) from centerpoint.

sgNicola, 2024-05-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

 The obstacle pointcloud-based validator node is incorrectly removing small DetectedObjects, such as traffic cones, identified by the centerpoint node. This leads to a misdetection.
[image]
[image]

### Expected behavior

The obstacle pointcloud-based validator node should accurately validate and retain all DetectedObjects recognized by the centerpoint node, including small objects like traffic cones.

### Actual behavior

The CenterPoint node detects small objects (e.g., traffic cones) with low frequency, but the obstacle point cloud-based validator node subsequently removes these detections.

### Steps to reproduce

1. Launch Autoware.Universe in an environment with traffic cones as potential obstacles.
2. Monitor the detection outputs from the centerpoint node and the subsequent validation by the obstacle pointcloud-based validator node.
3. Observe the failure of the validator node to retain valid DetectedObjects (traffic cones) identified initially by the centerpoint node.

### Versions

-OS: Ubuntu20.04
-ROS2: ROS2 galactic
-Autoware: Autoware.universe galactic

### Possible causes

1. Low Detection Frequency: The CenterPoint node may not be optimally configured or may lack the sensitivity needed to consistently detect small objects.
2. Validation Thresholds: The obstacle point cloud-based validator node may have stringent thresholds or criteria that do not favor the retention of small, infrequently detected objects.

meliketanrikulu, 2024-05-09:
@sgNicola  Thanks for your contribution.  Could you share example data to reproduce this problem.

sgNicola, 2024-05-09:
Hi, I have shared the ros2bag in this Google Drive link [provided.]
[https://drive.google.com/drive/folders/1CtjPF1jNsm2bRsyisEOX7EJDEtoroJnX?usp=drive_link](url)

meliketanrikulu, 2024-05-16:
> provided
@sgNicola I can not open data link . Could you check the link . Thank you

sgNicola, 2024-05-17:
@meliketanrikulu， I have checked the visitiproperty, could you please reopen the link? Thank you.

sgNicola, 2024-05-17:
Hi, @meliketanrikulu , I have checked the visiting property, could you please try again? Thank you.

amadeuszsz, 2024-07-08:
Hi @sgNicola 
Did you fix the issue already? If not, I would like to reproduce it but your link does not have rosbag file, I see only the meta file.

sgNicola, 2024-07-08:
Thank you for your attention, I have updated the link [
https://drive.google.com/drive/folders/1CtjPF1jNsm2bRsyisEOX7EJDEtoroJnX?usp=sharing].
Could you please check again?

amadeuszsz, 2024-07-10:
@sgNicola 
I reproduced your issue and here is brief summary:
1. Apart of fact the cone was detected as pedestrian, indeed `obstacle_pointcloud_based_validator` removes tiny objects.
2. This issue was addressed in past PR https://github.com/autowarefoundation/autoware.universe/pull/2894. Unfortunately for `galactic` branch users, this feature is not included.
3. Beside this dynamic number of points calculation, this detection is still being removed. I believe our default config gives us trade off between maximizing false positives / minimizing true positives removal. Users may play with settings to fit their scenarios. Based on distance and number of points (5~ in your example) I quickly calculated desired values - for your case you would need to change desired class `min_points_num` to 5 and  `min_points_and_distance_ratio` to 100 [here](https://github.com/autowarefoundation/autoware_launch/blob/main/autoware_launch/config/perception/object_recognition/detection/detected_object_validation/obstacle_pointcloud_based_validator.param.yaml).

[image]

I suggest to move on ROS 2 humble if it's possible and tune the config file, but keep that in mind you will end up with more false positives. I hope it will help you a lot!

sgNicola, 2024-07-12:
Thank you for your help!

amadeuszsz, 2024-07-24:
> Thank you for your help!

Closing as resolved 👍🏻

closed 2024-07-24 by amadeuszsz