autowarefoundation/autoware_universe #6940: Incorrect Detection of Roadside Trees as Obstacles on the Road (Autoware.universe+Carla 0.9.14 co-simulation)

sgNicola, 2024-05-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Where trees located beside roads are incorrectly considered as obstacles on the road. It disrupts normal vehicle operations by causing unnecessary avoidance maneuvers or stops.

### Expected behavior

The vehicle should go ahead.

### Actual behavior

The vehicle stops and does not proceed upon approaching a tree, mistaking it for an obstacle on the road.

### Steps to reproduce

1. Launch Autoware.Universe.
2. Drive the vehicle towards areas around the tree
3. Observe that the vehicle stops,  treating it as an obstacle on the road.

### Versions

-OS: Ubuntu 20.04
-ROS2: ros2 galactic
-Autoware: Autoware.universe galactic

### Possible causes

This behavior likely originates from the Surround Obstacle Checker module, which subscribes to the /perception/obstacle_segmentation/pointcloud topic. Furthermore, the absence of detected objects in the /perception/object_recognition/objects topic output may exacerbate the issue, indicating potential malfunctions in the perception module. 

### Additional context

This may have the same root cause as #6936.

sgNicola, 2024-05-09:
Hi, I would like to upload the ros2bag for steps to reproduce, however, it doesn't support db3 file type.
[image]
[image]
[image]

badai-nguyen, 2024-07-10:
@sgNicola FYI: I am working on kind of similar issue here https://github.com/autowarefoundation/autoware.universe/issues/5594

amadeuszsz, 2024-10-02:
@badai-nguyen 

Since #5594 is solved, can we close this issue? Does new feature handle these cases?

amadeuszsz, 2024-11-22:
Closing due to inactivity and issue might be already solved. Feel free to reopen this issue if needed.

closed 2024-11-22 by amadeuszsz

badai-nguyen, 2025-01-10:
> @badai-nguyen
> 
> Since #5594 is solved, can we close this issue? Does new feature handle these cases?

@amadeuszsz I appologize for missed your mention. 
I closed the issue https://github.com/autowarefoundation/autoware.universe/issues/5594  since all necessary changes for the task were merged.
However, I'm not sure the new filter can handle this iusse in simulation environment since new filter mostly denpends on the performance of yolox model for image semantic segmentation.