autowarefoundation/autoware_universe #6942: The vehicle collides with the obstacle at the corner while turning

Kim-mins, 2024-05-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla, and I found a situation that the ego vehicle collides with the obstacle at the corner while turning.

https://github.com/autowarefoundation/autoware.universe/assets/50267797/1b9176fd-6f62-4703-82b3-5bdf23f93b80

https://github.com/autowarefoundation/autoware.universe/assets/50267797/d996e4ed-c0f0-4b61-8b06-de2e7565b9d7

On rviz video, the points from the fence are detected and overlap with the edge of the drivable area, but the ego vehicle eventually collides with the obstacle.

### Expected behavior

No collision while turning

### Actual behavior

But the collision was made.

### Steps to reproduce

Here's a [[ros2bag file](https://drive.google.com/drive/folders/1F26Gz-S_zBBhaw8yVTGVrCd4MTdsEnpD?usp=sharing)] for the reproduction.

### Versions

- Autoware: [20240115, prebuilt docker](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/167294357?tag=humble-20240115-prebuilt-cuda)
- Carla: 0.9.13
- OS (host): Ubuntu 20.04
- OS (docker): Ubuntu 22.04

maxime-clem, 2024-06-14:
The main issue comes from bad control: the vehicle deviates significantly from the trajectory when colliding with the obstacle (~45cm of lateral error).

[image]

The control issue should be improved by the new CARLA interface (https://github.com/autowarefoundation/autoware.universe/pull/6859).

Kim-mins, 2024-07-11:
Sorry for late reply @maxime-clem.

Thank you for checking. 
I'll check it with the new Carla interface, and I'll reopen this if I got the same issue.

Thanks!

closed 2024-07-11 by Kim-mins