autowarefoundation/autoware_universe #6949: The ego vehicle ignores the NPC Vehicle at the intersection and makes collision.

Kim-mins, 2024-05-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team, 

I'm currently running Autoware with Carla, and I found a situation that the ego vehicle ignores the NPC Vehicle at the intersection, and eventually makes collision.

Here's the video of the situation: [[frontview](https://drive.google.com/file/d/18-0Ak5Kk1NSXwoK1B1Zal4FRsd-4XJT8/view?usp=sharing)] [[rviz](https://drive.google.com/file/d/1i63VVYuFnXKEPMD5klg74_xJtjG0MaVC/view?usp=sharing)]

At 0:10 of rviz video, the ego vehicle encounters a NPC Vehicle at the intersection.
However, even though the ego vehicle knows he need to stop(there's a red line with the phrase 'intersection'), the vehicle does not slow down, and they collides.

### Expected behavior

In my opinion, since the ego vehicle cannot know if other vehicles stop or not, the best policy is to stop and go after the NPC Vehicle passes.

### Actual behavior

The ego vehicle did not yield to the NPC Vehicle, and they eventually collides.

### Steps to reproduce

Here's a ros2bag file for the reproduction: [[ros2bag](https://drive.google.com/drive/folders/1zXFuF198OUlFcgE5yQvmY8or_jpfBUT9?usp=sharing)]

### Versions

- Autoware: [20240115, prebuilt docker](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/167294357?tag=humble-20240115-prebuilt-cuda)
- Carla: 0.9.13
- OS (host): Ubuntu 20.04
- OS (docker): Ubuntu 22.04

soblin, 2024-05-08:
It looks like the map has no "turn_direction" tag for the intersection lane. I can see it because the blinker is not turned on when the vehicle turns left.
Please see https://autowarefoundation.github.io/autoware.universe/main/planning/behavior_velocity_intersection_module/#requirementslimitations

Kim-mins, 2024-05-09:
Thank you for the response @soblin!

I should have written my settings in more detail. Sorry for that. 
To my understanding, the blinker always being turned on while turning if `turn_direction` tag is included  to the map, but actually [the bridge I'm using now](https://github.com/evshary/zenoh_carla_bridge) does not support the `turn_indicator` signal.. so maybe we cannot get any clue from blinker.

Here's my [lanelet2 map](https://drive.google.com/file/d/12NK195Bc38yVgolrGfiqoyJRniMeYqg8/view?usp=sharing), and I checked that the intersection has correct `turn_direction`s, and I remember that the directions are also represented in RViz. (like some arrows below on the intersection lane)
[image]
Also, if I remember correctly, the phrase 'intersection' does not show up when there's no `turn_direction` tag.