autowarefoundation/autoware_universe #6958: Replace serialized detected object mergers to reduce complexity

technolojin, 2024-05-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

The current object tracker consists of object mergers, trackers, and tracking-merger. However, there are several issues with the current process:
1. The merged object takes the worst latency of all detectors.
2. Introducing a new detector to the pipeline can be complicated.
3. It is difficult to inspect faulty object detection/association due to the complexity of the merge process.

Here is the perception diagram. The object merger is highlighted as a red box.
[image]

Based on [the current launcher](https://github.com/autowarefoundation/autoware.universe/blob/2f46b25b189e022077bc5e05bca2c77264004e28/launch/tier4_perception_launch/launch/object_recognition/detection/merger/camera_lidar_radar_merger.launch.xml), the structure inside is as following.

[image]

The object_association_merger is used 2 or 3 times. Depends on which detection mode is selected (option switches are marked as pink diamond), the input also changes. 

### Purpose

The purpose of this task is to:
- Mitigate the complexity of the perception launcher (perception pipeline configurations)
- Simplify the detection merge process, association process, and tracker existence management
- Reduce perception pipeline latency
- Provide redundancy in case of sensor/detection failure

### Possible approaches

To achieve the goal, the following components will be developed and implemented in the multi_object_tracker:
* Channel configuration: Provide all available sources of detections.
* Input manager: Manage multiple detected object message channels.
* Process timing algorithm: Operate the tracking process in an optimal set of objects to reduce latency.
* Existence model: Model the existence probability of each detection source to understand the contributions of detections.
* Debugger: Visualize association results and existence probability on each channel.

### Definition of done

- Enable the multi_object_tracker to subscribe to single or multiple detected objects.
- Ensure that it works the same way if the merged detection objects message is set as a single source.

meliketanrikulu, 2024-05-30:
Hello @technolojin . Can we close this issue. Your PR looks merged

technolojin, 2024-06-05:
@meliketanrikulu 
Thank you to have interest of this work!
I am working on a launcher refactoring so that the multi-input mot can be tested in the perception pipeline. I may link the work here.

closed 2024-06-12 by technolojin