autowarefoundation/autoware_universe #6962: Cannot go downhill even though there's no obstacle in front.

Kim-mins, 2024-05-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla, and I found a situation that the ego vehicle cannot go downhill even though there's no obstacle in front of him.

Here's a frontview: [[frontview](https://drive.google.com/file/d/1I-WyYTftjuurNMQYk6fgFM111H5ZD-mJ/view?usp=sharing)] and rviz video:

https://github.com/autowarefoundation/autoware.universe/assets/50267797/2d28ad93-efda-4ee9-89a9-c7a89df80c8c

At 0:11 of the video, the ground not filtered well is considered as an obstacle, and after that, the ego vehicle stops at the end of the downhill.

### Expected behavior

I hope the ego vehicle to drive downhill with no interference.

### Actual behavior

The ego vehicle cannot drive down the hill.

### Steps to reproduce

Here's the corresponding ros2bag file for the reproduction: [[ros2bag](https://drive.google.com/drive/folders/1KgaVbHHvJ4LviAjgcsDKmuD3k9xSK2JC?usp=sharing)]

### Versions

- Autoware: [20240115, prebuilt docker](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/167294357?tag=humble-20240115-prebuilt-cuda)
- Carla: 0.9.13
- OS (host): Ubuntu 20.04
- OS (docker): Ubuntu 22.04

### Possible causes

I think the ground perceived as an obstacle is a problem, but it ended in short term, and it was filtered anyway.
But I think the situation that, even after the ground is filtered well, the ego vehicle cannot move is the problem.

### Additional context

This issue could be similar to #6555, but I thought that the ground is not a sudden obstacle.

idorobotics, 2024-05-13:
#6936

brkay54, 2024-05-14:
Hi, we switched obstacle_cruise_planner recently ([here](https://github.com/autowarefoundation/autoware_launch/pull/765)) which is using the predicted objects rather than pointcloud. A newer version would solve the problem. However, I think we should improve the ground filtering in high-slope roads.

Kim-mins, 2024-05-29:
Sorry for late reply @sgNicola @brkay54.

I checked the PR and tried to run Autoware version [here](https://github.com/autowarefoundation/autoware.universe/tree/7877192cb5700c1db15109dcd959785fc381951d), which is much more recent, but it seems the problem still remains..

Here's the rviz video: [[rviz](https://drive.google.com/file/d/1fOXJswc9vNU_Vvkg0YGMmJIsroJFyZsW/view?usp=sharing)]

As you can see in the video, the ground is perceived as an obstacle while downing the hill, and it is clearly filtered at all.
However, after the filtering the vehicle does not move, which seems the same as the issue of the past version.

You can also reproduce the issue with ros2bag here: [[ros2bag](https://drive.google.com/drive/folders/1u2ieyN0HWVj7G_ZLGLZQ-OBNlknYi3it?usp=sharing)]

Thank you!

amadeuszsz, 2024-11-08:
Closing due to inactivity. Feel free to reopen if needed.

closed 2024-11-22 by amadeuszsz