autowarefoundation/autoware_universe #6968: The ego vehicle sometimes does not make lane change behind the NPC Vehicle.

Kim-mins, 2024-05-10:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla, and I met a situation that the ego vehicle sometimes does not make lane change behind the NPC Vehicle.

Here are the videos: [[frontview](https://drive.google.com/file/d/1jo0gAPkR8tPyQq_mKPULR45PcjJdESgk/view?usp=sharing)] and rviz below:

https://github.com/autowarefoundation/autoware.universe/assets/50267797/b10979de-e800-437d-813b-b023dc7a82e7

At the start of the rviz video, the trajectory for lane change instantly show up, but it disappears right away.
Also, at 0:03, the lane changing trajectory also appears, but only the straight trajectory remains.
So the ego vehicle stops forever behind the stopping vehicle.

However, I'm confused as I also observed sometimes the lane change is successful: [[frontview2](https://drive.google.com/file/d/1dbR-vWDuDXy3Nw_bf9EzOK8e8EDnQ1RF/view?usp=sharing)]
As you can see in the frontview2 video, every NPC Vehicles and the starting point of the ego vehicle are the same, but the ego could change lane and successfully goes to the destination.

### Expected behavior

I hope changing lane always to be successful.

### Actual behavior

But sometimes the ego vehicle cannot properly make lane change.

### Steps to reproduce

Here's the ros2bag file for the reproduction: [[ros2bag](https://drive.google.com/drive/folders/1sfPGDZzQuw8KRHyOLdCXZ_e6iuDuuco5?usp=sharing)]

### Versions

- Autoware: [20240115, prebuilt docker](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/167294357?tag=humble-20240115-prebuilt-cuda)
- Carla: 0.9.13
- OS (host): Ubuntu 20.04
- OS (docker): Ubuntu 22.04

TianYifan-Vincent, 2024-05-30:
@Kim-mins Sorry to bother you. I'm also learning on the scenario about the ego vehicle making lane change. I have tried the bridge [zenoh_carla_bridge](https://github.com/evshary/autoware_carla_launch) and [open_planner](https://github.com/ZATiTech/open_planner). But I can't reappear the lane changing scenario when the ego vehicle runs behind the NPC vehicle, even not lane change trajectory like what you presented, only the straight trajectory. Do I need to write additional code for this in the bridge or does my [lanelet2 map](https://drive.google.com/drive/folders/1GBcV9uXAq3QVrLfQdqI0LTIRKms_r8kl#/) not support lane changing?

Kim-mins, 2024-06-03:
Hi @TianYifan-Vincent!

For the lane change, you should modify the `lanelet2 map` file.
I referred to the map file provided [here](https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/), and modified the lanes around the starting point of the video above.

TianYifan-Vincent, 2024-06-05:
@Kim-mins Thanks for your help! Indeed this awsim map is more standardized and detailed. The lanelet2 map file can be used and modified in autoware, but how to create the corresponding Carla map?

Kim-mins, 2024-06-05:
Actually, I don't know how to create carla map either.. I just partially modified the carla map file referring to the lines for supporting lane change from awsim's map.

sgNicola, 2024-06-20:
> @Kim-mins Thanks for your help! Indeed this awsim map is more standardized and detailed. The lanelet2 map file can be used and modified in autoware, but how to create the corresponding Carla map?

Hi, you can use [RoadRunner](https://ww2.mathworks.cn/products/roadrunner.html) to create the corresponding Carla map.

Kim-mins, 2024-07-11:
Sorry for late response @sgNicola.

Thank you for the information! I'll try it soon!

equantumWang, 2024-12-15:
Hi guys I may find the solution of this proble, see issues below:
[https://github.com/orgs/autowarefoundation/discussions/4577](url)
[https://github.com/hatem-darweesh/op_agent/issues/3](url)

Kim-mins, 2024-12-16:
Thank you for sharing @equantumWang.

Actually, as I mentioned in [this comment](https://github.com/autowarefoundation/autoware.universe/issues/6968#issuecomment-2144369870), I aware that the map file does not support lane changing itself, so I already modified the map to support it.
This issue is regarding the lane changing functionality of Autoware, which does not support lane changing properly even when the map supports lane changing.

Thanks.