autowarefoundation/autoware_universe #6993: Improve Roll and Pitch estimation in Localization

mitsudome-r, 2024-05-13:


meliketanrikulu, 2024-05-24:
### Progress:
To understand the source of the problem, I performed the following tests one by one:
1. To make sure that EKF is the source of the problem:
  I ran the localization completely with NDT. I gave NDT output as initial pose to NDT and published the map to baselink transform with NDT poses. In this way, I completely removed EKF from the system. I compared these results with the default autoware results.
  You can see the results in the video [here](https://www.youtube.com/watch?v=ckPNYZCzxO0)

  Test Results: As seen in this video, the point cloud shakes up and down in the default autoware (NDT + EKF) when passing through the speed bump. This error does not occur when run with NDT only.

2. Simulation Environment Test: Ensuring that EKF inputs, sensor calibrations or any other vehicle specific problems are not causing the error.
   I created the same scenario in a simulation environment and reproduced the same problem with awsim_labs_sensor_kit and sample_vehicle.
   You can see the results in the video [here](https://www.youtube.com/watch?v=dl00aDIoNnM)
   Test Results: In the simulation environment, I also saw that errors occurred when passing over bumps. From here I came to the conclusion that any vehicle specific problem (calibration, input twist data ) is not the source of the problem.
At the same time, I visualized the changes in pitch angle when it passed through the speed bump, both in the simulation environment and with real life data.I observed similar behavior in both. You can see these results below: 
<div align="center">
[image]
</div>

### Next Steps: 
After making sure that EKF is only the source of the problem, I will try to solve the problem by debugging EKF in the simulation environment.

meliketanrikulu, 2024-06-26:
I tried to use the Roll and Pitch values ​​from the IMU directly in EKF. 
1. Simulation Test : 
The problem was completely solved with this method in the simulation environment:
Test video is [here](https://www.youtube.com/watch?v=bMwIv_a8RM8)

 NDT and EKF roll and pitch values gave very similar results in simulation environment:
[image]

2. Real World Test
In tests performed on the real vehicle, I saw that the results improved, but the problem was not completely solved. 
Test video is [here](https://www.youtube.com/watch?v=xXozVPgaYtQ) 

[image]

meliketanrikulu, 2024-07-02:
I tested EKF which is IMU added with 3D Distortion Corrector [PR](https://github.com/autowarefoundation/autoware.universe/pull/7137)
I've seen the results get better as a result. I achieved very good results when IMU added to EKF and 3D Distortion Corrector were used together. However, the problem is not completely solved yet. So I'll keep working on it
The test video link is [here](https://www.youtube.com/watch?v=VXDYWmPCpic)

meliketanrikulu, 2024-07-03:
I will create a PR for this issue using the results below as reference.
1. Simulation Test:
 In the tests carried out in the simulation environment, you can see below that using imu directly solves the roll, pitch, and position z in EKF instead of calculating them with the 1D Filter:
[image]
Buna ek olarak aşağıdaki videoları inceleyebilirsiniz:
Default Autoware test video is [here](https://www.youtube.com/watch?v=U3ByvJ8QQXE)
EKF with IMU Integration test video is [here](https://www.youtube.com/watch?v=bMwIv_a8RM8)
2. Real World Test:
Although the same level of performance cannot be achieved in real-world tests due to different factors, there is an improvement. It also seems that the results are further improved by using the [3D Distortion Corrector](https://github.com/autowarefoundation/autoware.universe/pull/7137), which has been newly merged into Autoware.You can find the results below:
[image]
Real World Test Video is [here](https://www.youtube.com/watch?v=VXDYWmPCpic&t=16s)

meliketanrikulu, 2024-07-23:
I created new PR for solving this problem . Tested with adding delay compensation to roll and pitch. 
https://github.com/autowarefoundation/autoware.universe/pull/8095

meliketanrikulu, 2024-08-07:
Related [PR](https://github.com/autowarefoundation/autoware.universe/pull/8095) is merged. I will close the issue

closed 2024-08-07 by meliketanrikulu