autowarefoundation/autoware_universe #7065: dynamic_obstacle_stop_module does not check obstacles behind EGO's vehicle

beyzanurkaya, 2024-05-19:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Even if objects behind the EGO vehicle are detected, they are not used in path planning. This can result in a vehicle following the EGO from behind and trying to overtake it, potentially causing a collision. Therefore, objects located behind the EGO vehicle and whose predicted paths intersect with the trajectory should be taken into consideration in _dynamic_obstacle_stop_module_

### Purpose

To prevent a potential collision and make autonomous driving safer.

### Possible approaches

Objects behind the EGO vehicle whose predicted path intersects with the EGO's trajectory should not be **filtered out** in the _dynamic_obstacle_stop_module_ and should be taken into consideration.

### Definition of done

As long as a vehicle following the EGO from behind and attempting to **overtake** remains within the detection area, the EGO vehicle will stop.

maxime-clem, 2024-05-20:
Thank you for opening this issue.
Currently the `dynamic_obstacle_stop` module does not use the predicted paths but uses a simple "constant heading + constant velocity" model to predict the future positions of the dynamic objects ([see the documentation page](https://autowarefoundation.github.io/autoware.universe/main/planning/behavior_velocity_dynamic_obstacle_stop_module/)).

I agree we may want to consider predicted paths of dynamic objects. How about adding two parameters to the `dynamic_obstacle_stop` module ? One parameter `ignore_objects_behind_ego`, and one parameter `use_predicted_paths`.

By the way, I am planning to move the `dynamic_obstacle_module` to the new `motion_velocity_planner` (still a work in progress, see https://github.com/autowarefoundation/autoware.universe/pull/7064) to allow more accurate collision detection.

beyzanurkaya, 2024-05-20:
Thanks for your comment @maxime-clem 
Yes, I am aware that dynamic_obstacle_stop_module does not use predicted_path. 

I am considering proceeding as follows:

- First of all, I am thinking of adding a lambda function to [filter_predicted_objects](https://github.com/autowarefoundation/autoware.universe/blob/33bb24c6976f517ac8a57dfe62b5a7907a0741fa/planning/behavior_velocity_dynamic_obstacle_stop_module/src/object_filtering.cpp#L33C66-L33C90) to take into account the objects coming from behind by adding a parameter named `(ignore_objects_behind_ego)`. So the objects behind the vehicle will also be included in the filtered_object, which is the output of this function.

- At this stage, I plan to add a parameter to `use_predicted_path` of objects and check the intersection of the EGO's trajectory with the predicted paths of the filtered objects.

What do you think?

Additionally, I agree with you. I think it would be more logical for the _dynamic_obstacle_stop_module_ to be inside the _motion_velocity_planner_.

maxime-clem, 2024-05-20:
Your plan sounds good thank you :+1: If you need any help with developing these new features let me know. I will also be happy to review the PR once it is ready.

closed 2024-09-22 by beyzanurkaya