autowarefoundation/autoware_universe #7080: Intersection module is not triggered for low speed NPC

ahmeddesokyebrahim, 2024-05-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Intersection module should be checking and preventing collision with upcoming vehicles in intersection areas.
It has been observed in one of the scenarios that intersection module is not triggered and not generating the intersection virtual wall while there is an NPC with velocity 1.25 m/s.

https://github.com/autowarefoundation/autoware.universe/assets/36835765/462df39e-4cb5-4dca-9bea-6fe64864fac2

This scenario was investigated more carefully in planning simulator, and it was observed that intersection modules is triggered while NPC is moving with 1.75 and 2.25 m/s. In these cases there was predicted path generated for the NPC. While in the case NPC was moving with 1.25 m/s, there was not predicted path and intersection module is triggered when the NPC is almost totally in intersection area.

https://www.youtube.com/watch?v=IldtqE4neNs

### Expected behavior

Intersection module is triggered with low speed NPC

### Actual behavior

Intersection module is not triggered with low speed NPC

### Steps to reproduce

1. Open scenario simulator
2. Use this scenario 
  [UC-NTR-003-0004_case1.zip](https://github.com/autowarefoundation/autoware.universe/files/15384003/UC-NTR-003-0004_case1.zip)
3.  Use this map
  [lanelet2_map.zip](https://github.com/autowarefoundation/autoware.universe/files/15383999/lanelet2_map.zip)

### Possible causes

A possible root cause of this issue is that ` min_velocity_for_map_based_prediction` is set to 1.39 m/s. 
Reducing this parameter to 1.0 for example may lead to generating predicted path for the NPC wit 1.25 and intersection module should be triggered

soblin, 2024-05-21:
Please lower this parameter
https://github.com/autowarefoundation/autoware_launch/blob/0aea8a35b9831e207827504297c6ac1a2e6cd9e1/autoware_launch/config/planning/scenario_planning/lane_driving/behavior_planning/behavior_velocity_planner/intersection.param.yaml#L105

ahmeddesokyebrahim, 2024-05-23:
> Please lower this parameter and see if eog reacts to NPC
> 
> https://github.com/autowarefoundation/autoware_launch/blob/0aea8a35b9831e207827504297c6ac1a2e6cd9e1/autoware_launch/config/planning/scenario_planning/lane_driving/behavior_planning/behavior_velocity_planner/intersection.param.yaml#L105

Hi @soblin -san

Thanks for your comment!

This is the scenario [output](https://youtu.be/vxJkyEiq3uQ) after changing [ignore_parked_vehicle_speed_threshold](https://github.com/autowarefoundation/autoware_launch/blob/0aea8a35b9831e207827504297c6ac1a2e6cd9e1/autoware_launch/config/planning/scenario_planning/lane_driving/behavior_planning/behavior_velocity_planner/intersection.param.yaml#L105) to 0.1 - which is should be less than the NPC speed and still the intersection module is not triggered.

Please have a look to this [PR](https://github.com/autowarefoundation/autoware_launch/pull/994) and check the scenario [output](https://youtu.be/Tm5I4pCEalQ) after the PR and let me know your opinion.

Thanks again.

ahmeddesokyebrahim, 2024-05-28:
Hi @soblin -san & @tkimura4 -san

I have a question regarding the parameter [min_velocity_for_map_based_prediction](https://github.com/autowarefoundation/autoware_launch/blob/b6a858a924829d8127f3dc3c1f3b93f21cfb2406/autoware_launch/config/perception/object_recognition/prediction/map_based_prediction.param.yaml#L7). Is there a specific reason that led to setting the parameter to `1.39` instead of any other less value ?

soblin, 2024-07-03:
@ahmeddesokyebrahim Is it inevitable to run the scenario with default configuration ?

ahmeddesokyebrahim, 2024-07-03:
> @ahmeddesokyebrahim Is it inevitable to run the scenario with default configuration ?

Thanks for your question @soblin -san. I am afraid that I am not totally getting the point of your question. Can you please clarify it more ? 
Would you like to have a different autoware_launch branch  to change the intended parameters for purpose of testing such scenarios ?

soblin, 2024-07-03:
I think that threshold is for ignoring noisy measurement from the tracker module in the real world. I don't mean to argue that that scenario should be changed to fit current value. But I also think that scenario is strange because the NPC is slowly passing the intersection although it(right turn) has priority over ego(straight) given the map context.

ahmeddesokyebrahim, 2024-07-09:
> But I also think that scenario is strange because the NPC is slowly passing the intersection although it(right turn) has priority over ego(straight) given the map context.

Thanks so much @soblin -san for your comment. I think it is a valid point and I will discuss it in more details with @brkay54  as he is the author of this scenario and leading the dense urban ODD scenarios creation activity. 
I will get back to you with the output of the discussion about feasibility/reality of such scenario.

brkay54, 2024-07-10:
Hi @soblin,

> I think that threshold is for ignoring noisy measurement from the tracker module in the real world.

I agree that the threshold helps in filtering out noisy measurements. However, a 5 kph threshold seems quite high for stopping objects. Do you think it would be possible to measure within a range of ±5 kph for stopping objects? In my opinion, we should consider reducing this threshold.

> But I also think that scenario is strange because the NPC is slowly passing the intersection although it(right turn) has priority over ego(straight) given the map context.

Actually, there is no right of way at this intersection. Our objective is to ensure that the ego vehicle should proceed if the NPC allows it to do so. Otherwise, the ego vehicle should wait to avoid a possible collision with the NPC.

soblin, 2024-07-16:
@ahmeddesokyebrahim @brkay54  All right ! Then I think we can lower the threshold in map_based_prediction module.

closed 2024-08-30 by YoshiRi