autowarefoundation/autoware_universe #7081: Incorrect implimentation in `planning/behavior_velocity_no_stopping_area_module`

veqcc, 2024-05-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

This is a dicussion in https://github.com/autowarefoundation/autoware.universe/pull/6986.

After https://github.com/autowarefoundation/autoware.universe/pull/6517, the variable `dist_from_area_sum` is not used anymore in `generateEgoNoStoppingAreaLanePolygon` function in `behavior_velocity_no_stopping_area_module` module.
See https://github.com/autowarefoundation/autoware.universe/blob/b0a0a3993c7c5c46e2d79a1b8cd3c27856bbf796/planning/behavior_velocity_no_stopping_area_module/src/scene_no_stopping_area.cpp#L331

In other words, the parameter `margin` is also unused, and in tern the following calculations seems incorrect:
https://github.com/autowarefoundation/autoware.universe/blob/b0a0a3993c7c5c46e2d79a1b8cd3c27856bbf796/planning/behavior_velocity_no_stopping_area_module/src/scene_no_stopping_area.cpp#L149C3-L160C43
```
  const auto & vi = planner_data_->vehicle_info_;
  const double margin = planner_param_.stop_line_margin;
  const double ego_space_in_front_of_stuck_vehicle =
    margin + vi.vehicle_length_m + planner_param_.stuck_vehicle_front_margin;
  const Polygon2d stuck_vehicle_detect_area = generateEgoNoStoppingAreaLanePolygon(
    *path, current_pose->pose, ego_space_in_front_of_stuck_vehicle,
    planner_param_.detection_area_length);
  const double ego_space_in_front_of_stop_line =
    margin + planner_param_.stop_margin + vi.rear_overhang_m;
  const Polygon2d stop_line_detect_area = generateEgoNoStoppingAreaLanePolygon(
    *path, current_pose->pose, ego_space_in_front_of_stop_line,
    planner_param_.detection_area_length);
```

Since the parameter `margin` is unused, the two variables `stuck_vehicle_detect_area` and `stop_line_detect_area` would be exactly the same values.

### Expected behavior

I do not know the expected behavior here, but I think the current implementation is different from what the maintainer expects.

### Actual behavior

I don't know.

### Steps to reproduce

If you run cppcheck, you can find `dist_from_area_sum` related warning.

veqcc, 2024-05-27:
@kosuke55 
When can you work on it?

kosuke55, 2024-05-31:
@veqcc @shmpwk 
thanks for your report and remind.
I discussed with @kaigohirao san and now working on this.

idorobotics, 2024-06-11:
Closed by https://github.com/autowarefoundation/autoware.universe/pull/6986

closed 2024-06-11 by idorobotics

kosuke55, 2024-06-14:
fixed by https://github.com/autowarefoundation/autoware.universe/pull/7280