autowarefoundation/autoware_universe #7124: Lazy detection of a static pedestrian makes collision.

Kim-mins, 2024-05-25:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla and I found a situation that lazy detection of a static pedestrian makes collision.
Here are the rviz and the front view videos: [[rviz](https://drive.google.com/file/d/1mL3NWwqJj4NMGH60kpBzBtsKDYfO7oJ1/view?usp=sharing)] [[frontview](https://drive.google.com/file/d/19O5WPguVPwl11oMcTFV41_uArZvK6wL5/view?usp=sharing)]

The rviz video shows that, Autoware detects the pedestrian only after the vehicle gets close, and eventually makes collision.

I also reported the similar issue #6709, but it seems the bug remains in that,
1. I'm running the latest version now(running the commit [7877192cb5700c1db15109dcd959785fc381951d](https://github.com/autowarefoundation/autoware.universe/tree/7877192cb5700c1db15109dcd959785fc381951d))
2. I also tried to apply `position_filter`, but the detection performance seems not changed.: [[rviz](https://drive.google.com/file/d/126LfNmUYifHYO9Mui_B5bQ9TxE7JcU0P/view?usp=sharing)].

### Expected behavior

I hope the pedestrian to be detected well and to make no collision.

### Actual behavior

But the pedestrian seems detected too slowly, and they eventually collide.

### Steps to reproduce

Here are the ros2bag files:
- with `lanelet_filter`: [[ros2bag](https://drive.google.com/drive/folders/1uz7KhiMVLuViQRwRbvOB3wFs6WvVF1VS?usp=sharing)]
- with `position_filter`: [[ros2bag](https://drive.google.com/drive/folders/1jrCeSdxs5GF5iCTK1Pe65M0XjCHuyX1W?usp=sharing)]

### Versions

- `autoware.universe`: [7877192cb5700c1db15109dcd959785fc381951d](https://github.com/autowarefoundation/autoware.universe/tree/7877192cb5700c1db15109dcd959785fc381951d)
- Carla: 0.9.15

YoshiRi, 2024-05-31:
@Kim-mins 
Thanks for your report. Sorry, I missed your issue due to lack of perception label.

I briefly checked your rosbag and found pointcloud number is too small.
Current clustering algorithm output clusters with more than 10 point and the result seems only have 2~4 points until vehicle approaches. The figure below shows the actual pointcloud number is 2 when distance is 30m.

[image]

You can try:
- increase lidar pointcloud by changing sensor settings. Use more dense LiDAR model.
- Fix `min_cluster_size: 10` parameter in `voxel_grid_euclidean_cluster.param.yaml` or `euclidean_cluster.param.yaml` to smaller number, such as  `min_cluster_size: 4`. Remember this may increase false positive.

Kim-mins, 2024-06-04:
No problem! Thank you for the response @YoshiRi!

I just tested by using more dense LiDAR model(double the density), and seems it works!
Here's the video: [[rviz](https://drive.google.com/file/d/15_ZBvbGy5aZdZeBrK296vRIkFbvZefFF/view?usp=sharing)]

Thank you for helping!

closed 2024-06-04 by Kim-mins