autowarefoundation/autoware_universe #7135: [intersection] After manual driving, the car will no longer stop due to other vehicles

kany-hsu, 2024-05-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

We observed vehicles will be ignored after we switch to manual driving in a "closed pattern path".

Here is a demonstration with the planning simulator:

https://github.com/autowarefoundation/autoware.universe/assets/170914373/f8c4f21a-5466-46a0-8982-a572abb20a97

At the beginning, we assigned a start point and destination, and ran it autonomously. The car would stop as we expected. After that, we switched to manual driving, drove the car back to the starting point, and assigned the destination again, but this time the car couldn't stop.

### Expected behavior

Intersection module detects other vehicles, and stops the car.

### Actual behavior

After switched to manual driving, drove the car back to the starting point, the car will no longer stop due to other vehicles.

### Steps to reproduce

1. Use the lanelet map below to run planning simulator.
[lanelet2_map.txt](https://github.com/autowarefoundation/autoware.universe/files/15451944/lanelet2_map.txt)
`ros2 launch autoware_launch planning_simulator.launch.xml map_path:=<lanelet_map_dir> vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit`
2. Do the simulation like the video.

### Versions

Ubuntu 22.04
The demonstration is run with the latest main.

### Possible causes

When we assigned the position of the car manually like 0:45-0:55 in `intersection.mp4`. It seems the old trajectory still exists, and it will be "extended" based on the new assigned positions of the car. When the old trajectory over the intersection, it somehow considered the car have passed the trajectory again. Then the variable below will be `true` which make the intersection module ignore other incoming vehicles.
https://github.com/autowarefoundation/autoware.universe/blob/cbaf62a36685b35110afc46097f9d5bba59f27fa/planning/behavior_velocity_intersection_module/src/scene_intersection.cpp#L1419

soblin, 2024-05-27:
Actually this phonemena is out of the scope of intersection module or behavior_velocity_planner. Please reset the goal and path again.

For intersection module's perspective, it seems that map has a very long "intersection lane". So **you have intentionally/unintentionally set a goal inside the intersection**  and you have not set a route that passes through the intersection.

kany-hsu, 2024-05-28:
> Actually this phonemena is out of the scope of intersection module or behavior_velocity_planner. Please reset the goal and path again.
> 
> For intersection module's perspective, it seems that map has a very long "intersection lane". So **you have intentionally/unintentionally set a goal inside the intersection** and you have not set a route that passes through the intersection.
> 
> Also you should shorten the intersection lanelet.

Thanks for your comment @soblin !

It works by resetting the goal again after the car arrived the old goal.

And now I think the current problem is that after the car arrives the goal, **the old route is not reset and still exists**. If we switch to manual driving at this time, the old route will be affected or "extended", leading to this phenomena unless a new route is manually set by resetting the goal.

For handling unexpected situations and debugging, we often need to switch to manual driving.

Is it possible to automatically reset the old route after the car arrive the goal?

maxime-clem, 2024-06-19:
> For handling unexpected situations and debugging, we often need to switch to manual driving.

You are correct and this should be supported. There was a bug with that feature but it is now fixed (https://github.com/autowarefoundation/autoware.universe/pull/7505).

> Is it possible to automatically reset the old route after the car arrive the goal?

Do you mean we should cancel the previous route once arriving at the goal ?

soblin, 2024-08-15:
Can we close the issue ?

kany-hsu, 2024-08-19:
Sorry for the late reply, I missed the message.
> Do you mean we should cancel the previous route once arriving at the goal ?

Yes, if we do what's shown in `intersection.mp4` from 0:45-0:55, the previous route will affect the judgment of the intersection module, causing `is_permanent_go` to become "true", which leads to the failure of the intersection module .
https://github.com/autowarefoundation/autoware.universe/blob/28170c3a87752b687344dda563672ae0a8ea598b/planning/behavior_velocity_planner/autoware_behavior_velocity_intersection_module/src/scene_intersection.cpp#L1409C5-L1409C29