autowarefoundation/autoware_universe #7161: 3D object tracking

beginningfan, 2024-05-29:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently, the tracking module in Autoware is 2D, which only considers the information of the xy plane and cannot adapt to situations such as uphill and downhill. This task aims to design a 3D tracking module to solve the problem of poor 2D performance.

### Purpose

Design a 3D tracker to improve tracking performance

### Possible approaches

Autoware also has a [bytetracker](https://github.com/autowarefoundation/autoware.universe/tree/main/perception/bytetrack). I will try the effect of this tracker. If it works well, I will transform this tracker into 3D. 

### Definition of done

- [ ] Testing the existing bytetracker
- [ ] Add a new 3D tracking package
- [ ] Input and output result adaptation to Autoware topics

beginningfan, 2024-07-05:
Sorry for the late update. I was unable to test bytetracker because the yolox module could not generate roi.
Today I tested the tracking effect of bytetracker. Even when facing fast-moving pedestrians or being temporarily blocked, it still has a relatively good tracking effect. ID switching will only occur when the target's movement direction changes significantly or disappears for a long time.
There is the test [video](https://github.com/autowarefoundation/autoware.universe/assets/103237402/a48e1e5d-9e12-4385-951c-33f87aae2b7a)
Next I will try to modify the tracking logic of bytetracker to 3D and add it to multi_object_tracker

beginningfan, 2024-07-24:
Basic tracking functions have been implemented, but there are still some problems. 
1. Information such as speed and heading angle has not been added. 
2. The truck cannot be tracked (for unknown reasons)

This is a test video. The blue bbox is the detected target and the green bbox is the tracked target.

https://github.com/user-attachments/assets/efc83b69-51da-4f7a-89d5-459b12c38794

alanmengg, 2024-08-21:
we have add uuid to track objects, and currently working on solving yaw problem.

https://github.com/user-attachments/assets/2b030322-ca2a-4b53-9cd7-ca99928ff1c5

alanmengg, 2024-08-28:
the yaw problem has been partially solved  in tracking process, as showed in picture below, orientation of turning object is correct.
[image]
still the object orientation will shake if the detection yaw is noisy, we will try to stabilize yaw tracking process.

alanmengg, 2024-09-04:
After frame-wise analysis, we discover that yaw is caused by centerpoint detection often give inconsistent yaw  between two frames, which has  about 180 degree difference. as showed below pitctures(track confidence is draw on left front point)
[image]
[image]
To resolve the problem, we think add ego velocity into tracking process which can estimation absolute velocity of track, then yaw estimation can be inferred from it.

alanmengg, 2024-09-18:
There are a lot of low confidence(<0.5) but real object which detected by centerpoint. 
To track those objects, we modified bytetrack a little by two frame objects association in initilization.
As showed in below video, the bus has been tracked.
https://github.com/user-attachments/assets/89f8e2e9-912d-4f44-bfd3-7524f047a681

cyn-liu, 2024-11-05:
> After frame-wise analysis, we discover that yaw is caused by centerpoint detection often give inconsistent yaw between two frames, which has about 180 degree difference. as showed below pitctures(track confidence is draw on left front point) ![Screenshot from 2024-09-04 11-51-04](https://private-user-images.githubusercontent.com/104343391/364217196-98f46af5-4d80-4ecb-b4b7-15438fece6bf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA3OTAyMTIsIm5iZiI6MTczMDc4OTkxMiwicGF0aCI6Ii8xMDQzNDMzOTEvMzY0MjE3MTk2LTk4ZjQ2YWY1LTRkODAtNGVjYi1iNGI3LTE1NDM4ZmVjZTZiZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNVQwNjU4MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZTQ0NzQ2OTNlNzdjNzA1OWU3ODM0MDA1YTFkMTdmNTFjNmU2MWMwODFkOWFjMmRjNTgxZWQ2Y2JhM2Q2M2I1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.BvfMubjkAcJJl0KB0maEzpDJJTTC_uJQgx_HJ59TTEk) ![Screenshot from 2024-09-04 11-50-54](https://private-user-images.githubusercontent.com/104343391/364217201-532d2f36-4e71-4111-9b80-7fcb9251ab47.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA3OTAyMTIsIm5iZiI6MTczMDc4OTkxMiwicGF0aCI6Ii8xMDQzNDMzOTEvMzY0MjE3MjAxLTUzMmQyZjM2LTRlNzEtNDExMS05YjgwLTdmY2I5MjUxYWI0Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNVQwNjU4MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NTZiYmI5YTA0ZjBmZmUzMDYwNTljMjM5MGFhMTk5ODgwYmFkZDU5MzViYTZkZjg2NmY5NjA3NGJjYmE5YmM4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.ZjXjzGzrWzxL3MLhqDPXMRU6Q54soSYuObk30hMQqUE) To resolve the problem, we think add ego velocity into tracking process which can estimation absolute velocity of track, then yaw estimation can be inferred from it.

If there is a 180 degree difference in the yaw of the same tracking target between two frames, the algorithm will correct the yaw based on the difference between the observed yaw and the predicted yaw each time the tracking trajectory is updated.