autowarefoundation/autoware_universe #7166: Misclassification of static obstacle into cyclist makes the vehicle stop.

Kim-mins, 2024-05-30:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla, and I found a situation that the misclassification of static obstacle into cyclist makes the vehicle stop.
Here's the rviz video:

https://github.com/autowarefoundation/autoware.universe/assets/50267797/e1b2a8d0-4e4a-4376-bd54-e35edced2c7b

According to the video, the vehicle does not move because of the static obstacle, which is not cyclist at all.
I checked that obstacle in Carla, referring to the [[frontview video](https://drive.google.com/file/d/17pwY98e9KuLvOx2b-J6l_GNOiRzAKIff/view?usp=sharing)], it was something other than the cyclist.
[image]

### Expected behavior

I hope the vehicle to drive as usual, classifying the static obstacle clearly.

### Actual behavior

However, the static obstacle is misclassified as a cyclist, which eventually make the vehicle stop.

### Steps to reproduce

Here's the ros2bag file for the reproduction: [[ros2bag](https://drive.google.com/drive/folders/1NpFsgB7aFSPf-ZkdRbv9M1ebp0OAvnLU?usp=sharing)]

### Versions

- autoware.universe: [7877192cb5700c1db15109dcd959785fc381951d](https://github.com/autowarefoundation/autoware.universe/tree/7877192cb5700c1db15109dcd959785fc381951d)
- Carla: 0.9.15

YoshiRi, 2024-05-31:
@Kim-mins 

I found this problem is caused by centerpoint detection. (sky blue bbox in figure below)

[image]

*Fig1. centerpoint detection in sky blue and clustering based detection in purple*

Though these solutions will be temporary one, you can try:
- Update LiDAR model to get high resolution pointclouds
- Change merger settings so that the autoware prioritizes clustering object than centerpoint

For example, you can fix `priority_mode` in `lidar_merger.launch.xml` 
```xml
  <!-- Merger -->
  <group>
    <!-- 1st merger to merge clustering + ML lidar detection-->
    <let name="merger/input/objects" value="$(var lidar_detection_model)/validation/objects" if="$(var use_validator)"/>
    <let name="merger/input/objects" value="$(var lidar_detection_model)/objects" unless="$(var use_validator)"/>
    <let name="without_dbt_and_filter" value="$(eval &quot;'$(var use_detection_by_tracker)'=='false' and '$(var use_object_filter)'=='false' &quot;)"/>
    <let name="merger/output/objects" value="temporary_merged_objects" unless="$(var without_dbt_and_filter)"/>
    <let name="merger/output/objects" value="$(var output/objects)" if="$(var without_dbt_and_filter)"/>
    <include file="$(find-pkg-share object_merger)/launch/object_association_merger.launch.xml">
      <arg name="input/object0" value="$(var merger/input/objects)"/>
      <arg name="input/object1" value="clustering/objects"/>
      <arg name="priority_mode" value="1"/>    <!-- change here from 0 to 1 so that merger will prioritize object1-->
      <arg name="output/object" value="$(var merger/output/objects)"/>
      <arg name="data_association_matrix_path" value="$(var object_recognition_detection_object_merger_data_association_matrix_param_path)"/>
      <arg name="distance_threshold_list_path" value="$(var object_recognition_detection_object_merger_distance_threshold_list_path)"/>
    </include>
  </group>
```

Kim-mins, 2024-06-04:
Thank you for the response and possible solutions @YoshiRi!

I tried your suggestions, but it seems they don't work..

For a resolution of the pointcloud, I increased the # of point per second from `1,200,000` to `10,000,000`(changed from carla's sensor setting), but it doesn't work. Here's the video: [[video](https://drive.google.com/file/d/1ubYkll2oSf3mJCRAmzwZKx1TtqnXsyoN/view?usp=sharing)]

I also changed the `priority_mode` to `1`(actually my default value was `2`, so I changed from `2` to `1`), but it seems the issue still remains. [[video](https://drive.google.com/file/d/1sIfY9PmamJxti0qrL3zOkAZoOM6IMdRi/view?usp=sharing)]

amadeuszsz, 2024-11-08:
Closing due to inactivity. Feel free to reopen if needed.

closed 2024-11-08 by amadeuszsz