autowarefoundation/autoware_universe #7205: Anbiguous `nullptr_t` in `common/autoware_auto_common`

veqcc, 2024-06-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The clang-tidy warns the following in https://github.com/autowarefoundation/autoware.universe/blame/main/common/autoware_auto_common/include/autoware_auto_common/helper_functions/message_adapters.hpp 
```
unknown type name 'nullptr_t'; did you mean 'std::nullptr_t'?
```

If there is any reason why you use `nullptr_t`, not `std::nullptr_t`, please explain it and avoid clang-tidy warnings!!

### Expected behavior

use `std::nullptr_t`

### Actual behavior

currently using `nullptr_t`

### Steps to reproduce

run clang-tidy

shmpwk, 2024-06-01:
@veqcc 
Thank you for your check. I agree to use `std::nullptr_t` insted of `nullptr_t`.

veqcc, 2024-06-01:
@shmpwk 
Do you create a PR by yourself? or, may I?

shmpwk, 2024-06-01:
OK I will!

closed 2024-06-02 by veqcc