autowarefoundation/autoware_universe #7210: Avoid `no viable conversion` warning in yabloc

veqcc, 2024-06-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Clang-tidy warns yabloc with the following in 7 modules:
```
no viable conversion from 'yabloc::common::Color' to 'std_msgs::msg::ColorRGBA_<std::allocator<void>>'
```
or
```
no viable conversion from 'common::Color' to 'std_msgs::msg::ColorRGBA_<std::allocator<void>>'
```
- https://github.com/autowarefoundation/autoware.universe/blob/6fe9fc648eaa3c1c36dc75fb8800212bd075bbfe/localization/yabloc/yabloc_particle_filter/src/common/particle_visualize_node.cpp#L96
- https://github.com/autowarefoundation/autoware.universe/blob/6fe9fc648eaa3c1c36dc75fb8800212bd075bbfe/localization/yabloc/yabloc_particle_filter/src/common/visualize.cpp#L48
- https://github.com/autowarefoundation/autoware.universe/blob/6fe9fc648eaa3c1c36dc75fb8800212bd075bbfe/localization/yabloc/yabloc_particle_filter/src/gnss_corrector/gnss_corrector_core.cpp#L139
- https://github.com/autowarefoundation/autoware.universe/blob/6fe9fc648eaa3c1c36dc75fb8800212bd075bbfe/localization/yabloc/yabloc_pose_initializer/src/camera/marker_module.cpp#L45
- https://github.com/autowarefoundation/autoware.universe/blob/6fe9fc648eaa3c1c36dc75fb8800212bd075bbfe/localization/yabloc/yabloc_common/src/ll2_decomposer/ll2_decomposer_core.cpp#L201
- https://github.com/autowarefoundation/autoware.universe/blob/6fe9fc648eaa3c1c36dc75fb8800212bd075bbfe/localization/yabloc/yabloc_common/src/ll2_decomposer/ll2_decomposer_core.cpp#L231
- https://github.com/autowarefoundation/autoware.universe/blob/6fe9fc648eaa3c1c36dc75fb8800212bd075bbfe/localization/yabloc/yabloc_particle_filter/src/ll2_cost_map/hierarchical_cost_map.cpp#L172

Could you fix the codes to avoid these warnings?

### Purpose

To reduce clang-tidy CRITICAL level warnings

### Possible approaches

nothing

### Definition of done

The warnings are reduced

KYabuuchi, 2024-06-03:
@veqcc Thank you for reporting the issue. I will try to fix it :+1:

KYabuuchi, 2024-06-07:
@veqcc  I have fixed the warning by  https://github.com/autowarefoundation/autoware.universe/pull/7299

closed 2024-06-07 by KYabuuchi