autowarefoundation/autoware_universe #7211: C++ bug in lambda capturing of structured bindings

veqcc, 2024-06-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

There is a bug in C++<=17 around lambda capturing for structured bindings: https://www.reddit.com/r/LLVM/comments/s0ykcj/why_does_clang_fail_with_error_reference_to_local/

This bug if fixed in C++20: https://releases.llvm.org/16.0.0/tools/clang/docs/ReleaseNotes.html#c-language-changes

Since Autoware runs with C++<=17, we have to avoid it.
Fortunately, clang-tidy makes the related warning!!
For example,
```
reference to local binding 'search_start_pose' declared in enclosing function 'behavior_path_planner::GoalSearcher::countObjectsToAvoid'
```

- search_start_pose : https://github.com/autowarefoundation/autoware.universe/blob/6fe9fc648eaa3c1c36dc75fb8800212bd075bbfe/planning/behavior_path_goal_planner_module/src/goal_searcher.cpp#L212
- search_end_pose : https://github.com/autowarefoundation/autoware.universe/blob/6fe9fc648eaa3c1c36dc75fb8800212bd075bbfe/planning/behavior_path_goal_planner_module/src/goal_searcher.cpp#L212
- object_exit_time: https://github.com/autowarefoundation/autoware.universe/blob/6fe9fc648eaa3c1c36dc75fb8800212bd075bbfe/planning/behavior_velocity_intersection_module/src/scene_intersection_collision.cpp#L269
- object_enter_time: https://github.com/autowarefoundation/autoware.universe/blob/6fe9fc648eaa3c1c36dc75fb8800212bd075bbfe/planning/behavior_velocity_intersection_module/src/scene_intersection_collision.cpp#L269
- max_lat_acc: https://github.com/autowarefoundation/autoware.universe/blob/6fe9fc648eaa3c1c36dc75fb8800212bd075bbfe/planning/behavior_path_lane_change_module/src/utils/utils.cpp#L95
- feasible_slow_down_vel: https://github.com/autowarefoundation/autoware.universe/blob/6fe9fc648eaa3c1c36dc75fb8800212bd075bbfe/planning/obstacle_cruise_planner/src/planner_interface.cpp#L557

### Expected behavior

Do not use lambda capturing for structured bindings.

### Actual behavior

nothing

### Steps to reproduce

nothing

soblin, 2024-06-01:
I referred to this article.(japanese)

https://onihusube.hatenablog.com/entry/2019/10/04/122001

IMO this is a trivial syntax sugar warning. Initial C++17 N4659 allowed the structured binding specifiers of a tuple to be referred from a lambda. But this was deprecated later because it was only allowed for a tuple-like object(our usage), not array/struct.

We are using g++ so there is no issue/bug. I want to keep using it because the code looks more tidy.

veqcc, 2024-06-01:
@soblin 
Thank for your quick investigation!

We should avoid clang-tidy CIRITICAL level warnings, even if it is actualy not a bug, in terms of maintainability.
( so that we can get useful infomation from clang-tidy results)
As you say, we can easily avoid the warning with a small tidy additional code.

closed 2024-07-22 by veqcc